kliment/Printrun #1385: How does pronterface access serial ports in macOS Ventura?

paulbeard, 2023-10-05:
Prusaslicer uses a CP102 USB/UART driver by name but Pronterface doesn't see that. Cura uses /dev/cu.usbserial0001 but Pronterface has no luck there. Not sure Cura does other than to install firmware: the command window doesn't see to access the M502/M500 commands I am trying to send. 

Pronterface sits at a Connecting…response forever. Is there a baud rate or other setting I need to change? Disconnecting/power cycling the printer yields a lot of debug 

2023-10-04 19:28:39,029 - Connecting...
2023-10-04 19:29:19,002 - Connecting...
2023-10-04 19:29:34,776 - >>> m502
2023-10-04 19:29:34,777 - Printer is not online.
2023-10-04 19:30:09,871 - Connecting...
2023-10-04 19:32:47,740 - Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 57600:
Maybe a bad baudrate?
2023-10-04 19:32:48,030 - Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 57600:
Maybe a bad baudrate?
2023-10-04 19:32:52,863 - Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 57600:
Maybe a bad baudrate?
2023-10-04 19:33:02,946 - Connecting...
2023-10-04 19:33:09,637 - Connecting...
2023-10-04 19:33:23,733 - Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 9600:
Maybe a bad baudrate?
2023-10-04 19:33:28,570 - Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 9600:
Maybe a bad baudrate?
2023-10-04 19:33:51,303 - Reset.
2023-10-04 19:36:06,235 - Not connected to printer.
2023-10-04 19:36:06,238 - Disconnecting from printer...
2023-10-04 19:36:06,238 - Exiting program. Goodbye!
2023-10-04 19:36:55,727 - Connecting...
2023-10-04 19:37:06,422 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:37:06,427 - Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 747, in _send
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

2023-10-04 19:37:06,427 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:37:06,428 - Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 747, in _send
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

2023-10-04 19:37:06,428 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:37:06,429 - Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 747, in _send
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

2023-10-04 19:37:06,429 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:37:06,430 - Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 747, in _send
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

2023-10-04 19:37:06,430 - Aborting connection attempt after 4 failed writes.
2023-10-04 19:37:06,430 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:37:29,908 - Not connected to printer.
2023-10-04 19:37:29,909 - Disconnecting from printer...
2023-10-04 19:37:29,909 - Exiting program. Goodbye!
2023-10-04 19:39:32,063 - Connecting...
2023-10-04 19:39:37,732 - >>> m502
2023-10-04 19:39:37,733 - Printer is not online.
2023-10-04 19:39:42,756 - Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
2023-10-04 19:39:42,761 - Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun/printcore.py", line 747, in _send
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

rockstorm101, 2023-10-05:
> Is there a baud rate or other setting I need to change?
> [...]
> Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 57600: Maybe a bad baudrate?
> [...]
> Got rubbish reply from /dev/cu.usbserial-0001 at baudrate 9600: Maybe a bad baudrate?

What is your printer model? Did 57600 and 9600 baud rates work with PrusaSlicer? My default go-to baud rate would actually be 115200 or 250000

paulbeard, 2023-10-05:
it is an anycubic mega S. I have never done anything w the baudrate or any
thing there.

paulbeard, 2023-10-06:
What should I see when Pronterface connects? A prompt? All I get is 
Connecting…

I have been able to get some connection (the printer signals that it has a connection audibly) w this: 
cu -I /dev/tty.usbserial-0001 -s 57600

but what should I see/what can I do to get something to happen?

rockstorm101, 2023-10-06:
> it is an anycubic mega S

A quick web search seems to suggest that the firmware for that printer model does not play nice with hosts like OctoPrint and (presumably) Pronterface (ref: https://github.com/OctoPrint/OctoPrint/issues/2309#issuecomment-356893095)

You could enable debugging (_Settings_ > _Debug communications_) and share your logs to confirm the issue. I would suggest you try connecting using baud rates 115200 and 250000.

> What should I see when Pronterface connects?

It will say "Connected." and then the jogging controls will no longer be grayed out and should allow you to move the extruder around.

I assume you are able to print from SD card using the touch screen, right? I mean, the printer is in working condition.

DivingDuck, 2023-10-06:
Pronterface should say in the Terminal window:
```
Connecting...
Printer is now online.
Using tool 0.
```
You can also activate -->Settings -->Debug Communications. This shows your printer temperatures for bed and extruder continuously.
```
SENT: M105
RECV: ok T:18.0 /0.0 @0 T1:18.0 /0.0 @0 B:17.6 /0.0 @0 B1:17.1 /0.0 @0
SENT: M105
RECV: ok T:17.9 /0.0 @0 T1:18.0 /0.0 @0 B:17.4 /0.0 @0 B1:16.8 /0.0 @0
SENT: M105
```
Can you please attach the Pronterface logfile (Printrun.log) and configuration file (printrunconf.ini)? You need to zip them. 
(Location for the log file: -->Settings -->Options -->User Interface -->Log path)

DivingDuck, 2023-10-06:
Oh, you was faster @rockstorm101 :)

DivingDuck, 2023-10-06:
Maybe one additional thing as I saw Cura mentioned:
Cure blocks the com port once it find a printer. When running Pronterface you need to close the Cura app because no other app is able to use the com port parallel to Cura. The only workaround is to disable the com port plugin in Cura, but then you can't do serial communications in Cura.

paulbeard, 2023-10-06:
This is all great information…

The machine as it sits in inoperable. It can sometimes accept commands from the screen, like printhead movements, but not reliably. It accepted a G28 command through Cura but not the M502 and M500 that are needed to restore it to factory defaults (or if it did, there was no feedback). Heating does not work so printing is no op. It was able to load/unload filament yesterday so printhead heating worked for a bit but bed heating does not. 

I did a firmware change and it was unresponsive after that, but I now have it restored to the OEM firmware and can make firmware changes at will. Previously it was unresponsive to any serial communications. Cura and Prusaslicer can both access the serial comms to make firmware changes and Cura will send some code instructions, but there is no feedback or confirmation in the "console" it provides. 

I never used pronterface before this so I have never seen it actually work. I will see about getting some of the debug logging later today. 

If there is a firmware that is known to work well on this machine, I'll try it. So far, nothing seems to work but it could be the machine itself. Not sure how something so simple internally could go so wrong but it seems like it is a box of compromises, not something designed to perform well over time.

rockstorm101, 2023-10-06:
> If there is a firmware that is known to work well on this machine, I'll try it.

I can only recommend [Marlin][1] which is the one I use on my printer with zero issues. However configuring it requires some deep understanding of your printer internals. Which is why you might want to have a look at [this project][2] which provides precompiled binaries for your printer brand. They've got a pretty [thorough wiki][3] to help select the right binary. Let us know if you get good results with it.

[1]: https://marlinfw.org/
[2]: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S
[3]: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)

paulbeard, 2023-10-06:
I have been exploring the Knutwurst project and I just discovered the more complete list of hex files that I have been looking for. I'll try flashing one of them tonight. Maybe this machine can be salvaged after all. It sits next to a Prusa so the comparisons are not favorable.

paulbeard, 2023-10-06:
Printrun doesn't have much to say. I flashed with one of the hexes I think is for this machine and didn't get much. I don't find a printrunconf.ini file anywhere. 

2023-10-06 07:24:31,353 - Connecting...
2023-10-06 07:24:58,221 - SENT: M105
2023-10-06 07:25:02,030 - SENT: M105
2023-10-06 07:25:05,850 - SENT: M105
2023-10-06 07:25:09,670 - SENT: M105
2023-10-06 07:25:13,481 - SENT: M105
2023-10-06 07:25:17,288 - SENT: M105
2023-10-06 07:25:21,092 - SENT: M105
2023-10-06 07:25:24,897 - SENT: M105
2023-10-06 07:25:28,699 - SENT: M105
2023-10-06 07:25:32,505 - SENT: M105
2023-10-06 07:25:36,313 - SENT: M105
2023-10-06 07:25:40,131 - SENT: M105
2023-10-06 07:25:43,933 - SENT: M105
2023-10-06 07:25:47,737 - SENT: M105
2023-10-06 07:25:51,551 - SENT: M105
2023-10-06 07:25:55,360 - SENT: M105
2023-10-06 07:25:59,175 - SENT: M105
2023-10-06 07:26:02,980 - SENT: M105
2023-10-06 07:26:05,760 - Connecting...
2023-10-06 07:26:05,935 - SENT: M105
2023-10-06 07:26:07,211 - Reset.
2023-10-06 07:26:09,741 - SENT: M105
2023-10-06 07:26:13,553 - SENT: M105
2023-10-06 07:26:17,358 - SENT: M105
2023-10-06 07:26:21,162 - SENT: M105
2023-10-06 07:26:22,643 - Not connected to printer.
2023-10-06 07:26:22,643 - Disconnecting from printer...
2023-10-06 07:26:22,643 - Exiting program. Goodbye!

DivingDuck, 2023-10-06:
It says a lot: I'm not able to connect.
Pls make a screenshot of -->Settings -->Options -->Printer settings

paulbeard, 2023-10-07:
[image: Screenshot 2023-10-06 at 4.59.22 PM.png]

paulbeard, 2023-10-07:
I dropped back to the OEM firmware just now, just hoping to get some
functionality out of this thing. Now I can't send the reset commands so I
can't actually run a job. Preheating doesn't work. I hear a clicking sound
as if a motor is stuck. And since I have the chassis open I can see that
the red light on the back of the display board is blinking. Any idea what
that means? I did have to remove the daughterboard that runs the display to
get to the serial connection when I was thinking I had to reflash the
bootloader. I assume if I didn't put it back right the display wouldn't
work and it does.

I do see a loose two pin connector (green/purple) that I don't remember
unplugging…I'll see if I can find some image that tells me where it should
be. It comes from the bundle of wires that connects to the X steppers. The
jog commands for X, Y, and Z all work depending on the firmware that's
loaded. So maybe it never was connected.

I know this is all hardware, not software/firmware, but maybe a decent
connection would help me diagnose some of this. If ever a machine needed a
"No User Serviceable Parts Inside" sticker, this is it.

DivingDuck, 2023-10-07:
I don't own an Anycubic mega S, so here is little to help you on this specific hardware/firmware problem. Maybe there is one who can help on this as they sold a lot of this printers around the world.

I think we should leave this thread open so that you have time to fix the hardware / firmware issues first. I mark this thread as "Not a bug" for now.

Maybe some additional hints in general:

- You may need a device driver software for your operating system. This usually came with the printer (manufacturer SD card?)

- You need to check whether your system acknowledge the printer as device or not. For macOS you can use a terminal window to check this: 

- Disconnect the printer from your computer and then open the terminal window and type `ls /dev/{tty,cu}.*` followed by [enter].
- Make a screenshot of the report.
- Then connect the printer via USB, wit a little and type `ls /dev/{tty,cu}.*` followed by [enter] once more. 
- Compare the two reports. You will find a new device connected to a serial port if the device was recognize by your computer.
- In case there is no connection you need to find out if the problem is related to the USB cable or your printer. I often recognize that people try to use extended USB cable with more than 3 or 5 meters and then had those issues because there a using in addition the printer in an harsh environment with lot of electromagnetic interference. Check if a another (shorter) cable can  solve the problem. Also check if another serial port of your computer works better.

For Pronterface, as already mentioned from @rockstorm101 your printer type is usually connected with a 115200 or 250000 baud communication. You can use a terminal tool like PuTTY as well to check the connection. Here you will find some informations about it: https://www.ssh.com/academy/ssh/putty/mac
In addition I attach an  example of my connection setup (Windows version of PuTTY):

[image]

And here how this is set up for Pronterface:

[image]

For selection the serial Port in Pronterface connect the printer to your computer and start pronterface. Then click on button Port. Usually this is enough to select your correct serial com port as this will refresh the next button right with the acknowledged printer on your computer. Then you need to choose the correct printer if there is more then one recognized. The last point is to select the baud rate as mentioned already. Here it is how i looks for my printer:

[image]

Hope this will help little bit. Feel free to ask if you have further questions. You are welcome doing so. And keep us informed if you was able to solve your hardware / firmware issues.

paulbeard, 2023-10-07:
Thanks so much for this and your patience. I think, after looking at a lot
of pictures on this machine's guts, that I know where that connection
terminates. I really can't fathom why they hotglued that pair of wires to
the daugherboard for the display but that was the clue that helped me see
it. Everything else is straightforward and makes it clear that this should
have worked all along, though I didn't try using ssh as a terminal app like
this…I always have 2-3 ssh sessions going so I can try that as well. I may
just get it back in action so I can get rid of it with the knowledge it
won't end up in the waste stream and get a better one, as they are getting
less expensive and more capable every day.

paulbeard, 2023-10-08:
Well, sometimes being too dumb to know when you're licked pays off…I think this machine is working again, as well as it ever did. Cura can connect and send commands: G28 is my go to, as it will jog the printhead enough to let me know it got through. The heating is working, due to my guess about where those two wires went. Getting ready to run the test file to see how well it works. If this works, I'd be up to try a firmware upgrade but I have tried so many. I need to be sure it's the right one this time. Pronterface is connected. 

[image]

For some reason, the machine seems to just halt when cura is on the serial port. Heating doesn't increase, nothing runs.I unplugged the cable and the machine picked up again. Pronterface doesn't have that effect. 

But it is having a lot of trouble printing, major adhesion issues that are outside the scope of what we're talking about. I scrubbed the build plate w soap and water and now I am seeing results. A firmware that supports mesh leveling would be favorite.

DivingDuck, 2023-10-08:
Hi,
good to hear your printer works again.

> For some reason, the machine seems to just halt when cura is on the serial port. Heating doesn't increase, nothing runs.I unplugged the cable and the machine picked up again. Pronterface doesn't have that effect.

As I wrote before:

> Maybe one additional thing as I saw Cura mentioned:
> Cura blocks the com port once it find a printer. When running Pronterface you need to close the Cura app because no other app is able to use the com port parallel to Cura. The only workaround is to disable the com port plugin in Cura, but then you can't do serial communications in Cura.

You can't run Cura and Pronterface at the same time. This is exactly the message you will receive when two apps try to communicate to the same serial port. 
Side note: Be also careful when you have establish more then one serial connection. When Cura starts it will scan all available serial ports and flood them with communication traffic witch cause other applications with communication errors and broken connections (e.g. running micro controller boards via USB connection at the same time).

Cleaning the build plate with hot water and dish soap is perfect. Rinse the bed with clear water and use a clean kitchen towel (don't wash these towels with softener). Avoid touching the print surface with your fingers and don't use isopropanol for cleaning as often mentioned a lot in forums. While it works fine for some bed materials like Garolyte / FR4 (as long as there is no grease e.g. from finger tips involved), but you will mostly spread the grease of an contaminated bed over the surface as a very thin layer. After the isopropanol is evaporated it will leave the grease back. Sooner or later you will loose the adhesion - mostly sooner...
Use a cheap glue stick like UHU, Pritt or what ever is available in your country. Write an "O" or a "X" on the print surface (only a little is needed) and wipe it evenly with a very wet paper towel  over the complete build plate so that the glue is evenly averaged over the complete build plate as a very thin layer. This is enough for multiple prints. The glue acts as an adhesion promoter under heat and release the printed part after cooling from the bed. Heat up the bed afterwards so that the water can evaporate.

Mesh leveling can help a lot, especially if the bed isn't even or you are a beginner. Knowing how to level a bed manually is something you need to practice a bit, but it is easy to do and take for my printers only a minute. I use stripes of use print paper scrap for doing this. Heat up the bed to the needed print temperature (the heat should be stabilized) and the nozzle to around 160°C, clean the nozzle if needed and make the 5 point check. Usually after doing this the second time the bed is leveled. For me this is faster then doing a complete mesh bed leveling - but if I had to run a printer farm I would use mesh leveling too :)

paulbeard, 2023-10-08:
Yes, I know you can't have two conversations on the same port. What is
happening is that Cura being connected stops the printer, as if it's
waiting for instructions. Pronterface doesn't have the same effect. It does
have some problems, as noted in the screenshot above. Any idea what that
means?

The newly revived machine has been working overnight (it is quite loud, I
now remember), 9 hours into an 11+ hour job. Looks good so far. Once it's
done. I am game to try a firmware update but I have to make sure I get the
right one. The ones from Knutwurst are so specific, down to the board rev
and screen type.

DivingDuck, 2023-10-08:
> What is happening is that Cura being connected stops the printer, as if it's waiting for instructions.

That's what happen when Cura try to connect to a printer. Roughly explained Cura try to establish a connection by initializing a serial connection and then try to find out what baud rate is used. Doing so it sent a gcode command to the printer with different baud rates and look for a specific answer. In case it do not become the expected answer it try again with an other baud rate for all known rates. I'm not quite sure how often it try to start the loop again but I guess they do repeat it a couple of time to be sure that there is no connection. (Think back to the time where we had use a acoustic coupler modem for communication between computers, It is more or less the same process and you could listen to this process back in these days). This behavior can force a printer to do "silly" things. 
Side note: Keep in mind that UltiMaker do not support the direct serial communication via Cura since quite some time. This is why users switch to other solutions like Pronterface

Pronterface is different in this behave. It don't try to establish a connection itself. The user have to do that manually and have to tell Pronterface the correct com port and baud rate. If the setup is wrong you only see that there is no connection possible and then the program waits for the user for correcting the setup and for starting the next connection attempt.

paulbeard, 2023-10-08:
Not cool, Cura…I know I get the annoying startup tune as soon as a serial connection is made, so this all makes sense, thank you. I'd love find the GCODE to turn that off. 

The machine just completed a 14:28 hour job that almost maxed out the Z axis and it looks good. It seems to be working better than when I unboxed it. ¯\_(ツ)_/¯ 
[image]

I may look into replacing that Rolls Royce turbojet fan before I worry about anything else, to be honest. I need to look more closely into what a firmware upgrade would do: mesh leveling would be useful, for example. I like how the Prusa does that and how it lays out a short run of filament near (0,0) as a priming exercise. I know that can be done in GCODE. Still no idea why it was "bricked" as a result of the firmware update but I inclined to think the OEM firmware or hardware doesn't handle serial comms properly.

DivingDuck, 2024-06-06:
@paulbeard,

was you able to solve the problem with your printer? I wonder if we can close this issue.

Meanwhile we have released a new Pronterface version you can checkout too if you want. https://github.com/kliment/Printrun/releases/tag/printrun-2.1.0
Best regards,
DD

DivingDuck, 2024-10-20:
There is no need to leave this open as the issue is no bug.

closed 2024-10-20 by DivingDuck