kliment/Printrun #1392: UI hangs after some time, print continues

VanessaE, 2023-11-21:
I'm using Pronterface 2.0.0rc8 from Debian Bookworm repository, desktop environment is XFCE 4.18.

I am getting frequent UI hang-ups.  The UI just stops producing output and stops accepting input.  When in this state, the window will completely lose its contents if, for example, I minimize it or roll it up and then restore it.  Said contents get replaced with a copy of whatever was under the window at the moment, mimicking the classic signs of a crashed program...

Except the program isn't totally deadlocked -- if there's a print going, it continues unabated, or at least so far as I've seen.  The condition of the UI seems to have no bearing on that.

It is not required to even have a print going.  Today, for example, I had it hang-up while the printer was doing a hotend temperature auto-tune.

If I try to close the hung program via the window close [X] icon, my click is ignored by Pronterface, but after a couple of seconds, my window manager detects this and offers to terminate the program, which always succeeds.

I can't see any pattern to this behavior.  It just...randomly dies.  It seems to be most prone to it if I simply leave it alone, but that could just because it spends most of its time going about its business or just sitting idle.

VanessaE, 2023-11-21:
I just noticed that when the UI hangs, I get this in the controlling terminal:
```
Attempted to write invalid text to console, which could be due to an invalid baudrate
```
It just floods and floods until I kill the program.

DivingDuck, 2023-11-21:
Hi @VanessaE 
are you able to run Pronterface from source ([Running from source](https://github.com/kliment/Printrun#running-from-source))? There will be no update for this ancient version. We had a lot of changes after this release candidate was build. I wonder if this happen with our actual code too. We are happy to help you to set up the code if needed.
(I don't dare to ask if your baud rate was set properly)  :blush:

VanessaE, 2023-11-21:
> Hi @VanessaE are you able to run Pronterface from source ([Running from source](https://github.com/kliment/Printrun#running-from-source))? 

No, last time I tried to build it, things did not go well.  See #1390, which you already commented on. :smile: 

> There will be no update for this ancient version.

Nor should there be, 2.0.0rc8 is horribly outdated.  Instead, I would hope the Debian package would get updated to something more recent.  There is a newer build (2.0.1) in Debian Sid/unstable, but @rockstorm101 said it's not ready to be pushed to stable due to probs with Pyglet (https://github.com/kliment/Printrun/issues/1390#issuecomment-1792236269).

> I don't dare to ask if your baud rate was set properly 😊

Well I wouldn't expect baud rate would apply when writing to a local terminal or console. :stuck_out_tongue: