kliment/Printrun #1394: Pronterface - Errno  6  Device not configured

wizwho, 2023-12-01:
Artillery X3 response after "Connected" message lasting a few seconds. Works fine on the X2.

How to fix?

rockstorm101, 2023-12-10:
HI @wizwho, what's your OS and what version of Pronterface are you running? I'm unable to obtain this error but my guess is that something is wrong with the serial port connection. Did you make sure no other app is trying to use the same port at the same time?

wizwho, 2023-12-10:
You are too late. I’ve already requested return and refund.

fyi: Pronterface works fine on my other printers, including the X2 which I have put back in service. My OS is Mac 14.1.2.

For more info on other reasons why I am so disappointed, even if I could get Pronterface working, review my Customer Service dialogs. 

Ted King

rockstorm101, 2023-12-11:
Do I understand correctly that this was a hardware issue then and not directly related to Pronterface? Could we close this issue then?

wizwho, 2023-12-11:
Yes, more likely a firmware issue.

rockstorm101, 2023-12-15:
Thanks for your quick responses. Closing then.

closed 2023-12-15 by rockstorm101