kliment/Printrun #1398: My Laser uses the RS 232C communication protocol and not G-Code

Dalinaffeti, 2023-12-25:
Can you please help me connect my laser to Pronterface. And if i change the function written in Pronterface, is there anything that i should be aware of? This is will be not an easy task so i would appreciate every help

DivingDuck, 2023-12-25:
Hi,
G-Code is not a communication protocol. You possibly want a USB to RS232C adapter for connecting your laser with your pc?

Dalinaffeti, 2023-12-25:
I have Pronterface on an RaspberyPi4 my goal is the automation or the control of the different 3D Printer elements throw Pronterface. The Laser that i am using is called YLP-HP-A which uses the DB25 cable.

rockstorm101, 2023-12-26:
> I have Pronterface on an RaspberyPi4 my goal is the automation or the control of the different 3D Printer elements throw Pronterface. The Laser that i am using is called YLP-HP-A which uses the DB25 cable.

 * Is this "YLP-HP-A Laser" installed onto a laser cutter machine? I mean, at a quick glance, that's only the laser, you would need motors and stuff to move the laser around and cut/engrave things. The G-code commands would control both motor movement and laser pulses.
 * Are you able to connect the Raspberry to the laser cutter? I would expect that connection to be done via USB.
 * Does your laser cutter understand G-code? Which firmware does it run?

Dalinaffeti, 2023-12-26:
> * Is this "YLP-HP-A Laser" installed onto a laser cutter machine? I mean, at a quick glance, that's only the laser, you would need motors and stuff to move the laser around and cut/engrave things. [...]

The laser doesnt move the Motors (Snapmaker F350) move the plateau, where the object will be build on. Its called additive process. after the laser hits the powder the object will be build level to level.

I contacte the Manufacturer and these is there respond: "There is no control via Ethernet for these older lasers. The lasers are usually controlled via the DB25 interface (e.g. via a programmable control card)."

and provided a document for me
[90807-spec E27344 R00 YLP-HP digital interface specification type YLP-HP-A (1).pdf](https://github.com/kliment/Printrun/files/13771431/90807-spec.E27344.R00.YLP-HP.digital.interface.specification.type.YLP-HP-A.1.pdf)

Dalinaffeti, 2023-12-26:
> This looks like a powerful fiber laser... https://photonlexicon.com/forums/showthread.php/20828-How-to-communicate-with-YLP-series-IPG-laser

Exactly its a powerful and a dangerous laser, therefore i cant be testing around with it. And i am short on knowledge to be honest. I am new to all of this

kliment, 2023-12-26:
Hello @Dalinaffeti

You are building a laser cutter out of two incompatible closed source components. I'm sorry to say it so directly, but this will not work the way you imagine it. Because the laser pulses and the movement have to be synchronized, the EM signal of the laser has to be controlled by the printer firmware, and not by the Pi, otherwise you will shoot the laser in places where you don't want to. You will not be able to talk to the laser controller from pronterface (or any other printer control application). You will need to implement support for the laser synchronization somehow in the firmware. Because the Snapmaker firmware supports laser cutting, it should be possible to get this synchronization pulse from the Snapmaker electronics, but this is something that might require additional hardware or the snapmaker manufacturer's permission and support. Once you have the emission modulation signal, you can control the remaining functionality of the laser (power level setting, safety features, guide beam, sequencing) using raspberry pi GPIOs, but this is not something you can do with pronterface - you will need to implement a piece of software and a driver to do all this. With a laser as dangerous as this you also need to think about safety interlocks which need to be implemented in hardware. If you're new to all this, it's probably not the best project to be starting with because of the danger involved.

kliment, 2024-01-03:
I'm sorry to have to tell you this so harshly, but if that's the case, then it will probably fail. If you are not comfortable and experienced with writing drivers for unfamiliar hardware and implementing hardware interfaces then this is a significantly too complex project for a bachelor thesis. To put it simply, the laser does not speak the same language as pronterface, and there's no way to teach it a different language. You will need to write a driver for it, that does speak the same language, and then implement an interface that can be accessed from Python, and then call that from Pronterface. You will need to implement functionality for setting the laser power, enabling the laser, and handle all the safety interlocks. After that, you can drive the laser beam timing from Klipper using a configuration like https://www.klipper3d.org/Using_PWM_Tools.html 

However, I want to make this absolutely clear: **We will not do your homework for you!** We're not going to implement new functionality just so that you don't fail your bachelor thesis. If you have chosen a topic that is not manageable within the scale of a bachelor thesis, then you need to take responsibility for that.

I'll give you some pointers to work from, but this is not an issue we can help you with:

1. The 25-pin digital interface can be controlled from raspberry pi GPIO. You can use something like the gpiozero library (https://gpiozero.readthedocs.io/en/latest/) to talk to it. All the control pins can be configured as gpiozero LEDs and all the diagnostic pins as gpiozero buttons. You can write a program that checks for errors, configures the laser power, and enables the beam, so that you can switch the emission on and off using the emission modulation signal. This is not a part of pronterface, but it's possible to add a custom button and make it call this new program you write. You will also need another program (and another custom button) to disable the laser after the cutting task is complete.
2. See https://www.klipper3d.org/Using_PWM_Tools.html and https://klipper.discourse.group/t/testers-wanted-laser-support-fast-pwm-updates/1771 for help with controlling the laser synchronization with Klipper. This would also require a connection between the printer electronics and the 25-pin connector (you need to connect the emission modulation signal and ground to the printer electronics).
3. You will have to modify the gcode files to include commands to enable/disable the laser at the correct moment in the movement sequence.

This is as much as we can do for you here - you're on your own from now. I'll close this issue now.

closed 2024-01-03 by kliment

Dalinaffeti, 2024-01-03:
Thank you for sharing your knowledge with me.