kliment/Printrun #1401: Running Pronterface on Aarch64 and Trying Other Tools in the Suite

silver2row, 2024-01-01:
Hello,

```
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 30, in <module>
    from printrun.pronterface import PronterApp
  File "/usr/lib/python3/dist-packages/printrun/pronterface.py", line 32, in <module>
    from . import pronsole
  File "/usr/lib/python3/dist-packages/printrun/pronsole.py", line 35, in <module>
    from . import printcore
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 45, in <module>
    from printrun.plugins import PRINTCORE_HANDLER
ImportError: cannot import name 'PRINTCORE_HANDLER' from 'printrun.plugins' (unknown location)
```

I have put the exported files in .bashrc and then sourced the .bashrc file to run my commands. This has not worked from my ~/$HOME directory.

This is fine. I can use `python3 /usr/bin/pronterface` to handle the issue. The above set of displaced imports in Python3 persist. 

Has anyone come to get this error and found a workaround yet?

Seth

P.S. Oh and should printrun work w/ aarch64?

rockstorm101, 2024-01-01:
> I have put the exported files in .bashrc and then sourced the .bashrc file to run my commands. This has not worked from my ~/$HOME directory.

Hi @silver2row, could you please share:
 1. your OS
 2. your version of Pronterface
 3. and how you installed Pronterface?
 
I don't really understand the process you describe above for installation. Did you try [installing via PyPI][1]?

[1]: https://github.com/kliment/Printrun#printrun-package-from-pypi

> P.S. Oh and should printrun work w/ aarch64?

Absolutely, it should work with no issues on ARM64.

silver2row, 2024-01-01:
1. Debian Bookworm
2. the Debian Bookworm install via `sudo apt install printrun`
3. `sudo apt install printrun`

I have not tried w/ PyPI b/c of Debian Bookworm not playing nice w/ `python3 -m pip install printrun` or any pip3 installs.

rockstorm101, 2024-01-01:
> 2\. the Debian Bookworm install via `sudo apt install printrun`

That's weird, that should work out of the box. Am I right in thinking that you can't just run `pronterface` in a terminal and get it to run? You are having to run it like `python3 /usr/bin/pronterface`? If so, that sounds like you PATH is not properly configured. Anyway, you seem to be missing the `plugins` directory for Printrun. The following command should show this output (or similar). A missing `plugins` directory there means the `apt install printrun` failed to install all files somehow?

```shell
$ ls /usr/lib/python3/dist-packages/printrun/plugins
__init__.py  __pycache__  sample.py
```

> I have not tried w/ PyPI b/c of Debian Bookworm not playing nice w/ `python3 -m pip install printrun` or any pip3 installs.

Well done. That's why we recommend using [Python virtual environments][1] instead of the "system's" pip install.

[1]: https://docs.python.org/3/tutorial/venv

silver2row, 2024-01-01:
oh...right. I cannot run pronterface and have it run. I am on a vncserver w/ vnc viewer while running the `pronterface` command.

I tried w/ `pronsole` and ended up w/ the same error.

I will get back to you on the `ls` command so that I can give some more insight on this issue. 

And okay about the virtual env. I will test it that way too. Thank you for dealing w/ me so far.

silver2row, 2024-01-01:
```
ls: cannot access '/usr/lib/python3/dist-packages/printrun/plugins/__init__.py': Permission denied
ls: cannot access '/usr/lib/python3/dist-packages/printrun/plugins/__pycache__': Permission denied
ls: cannot access '/usr/lib/python3/dist-packages/printrun/plugins/sample.py': Permission denied
```

silver2row, 2024-01-02:
Hello...I just did not want to use sudo or get into a root account. I am forced to do so and this solves the issue. I am sorry to give you trouble w/ these bunch of errors.

closed 2024-01-02 by silver2row