kliment/Printrun #1404: Moved the custom buttons over too far, and now i have broke the whole program

messy117, 2024-01-10:
/
[image]

i was moving my 3 custom buttons around on the GUI and i kept hitting 'move right -->' and i went to far, and it caused the program to crash. Now i cant open it to connect to my printer.

DivingDuck, 2024-01-10:
Hi @messy117,

I can't replicate this error. Moving a custom button to left or right is deactivated as soon you are on the last possible position. But it is obviously that you was able to screw up something in Pronterface. My first guess is a malformed file called printrunconf.ini. You can make a backup of this file and then delete it. Pronterface should then start again but w/o any setup/customization you did before.

Please provide us with some more information:

1. Operating System
2. Pronterface version
3. Please attach the following files (make a zip file of all before attaching)
The location for a Windows system is:
`c:\Users\YOUR_USER_NAME\printrunconf.ini`
`c:\Users\YOUR_USER_NAME\Printrun.log`
and if available the complete folder 
`c:\Users\YOUR_USER_NAME\AppData\Local\Printrun\`

This can help us to find out what happen wrong with your configuration. Thanks.

neofelis2X, 2024-01-10:
Hi @messy117,
I remember this bug, I had it myself a while back. This is a duplicate of #1328, which we were able to fix with PR #1331. Unfortunately, this fix has not yet been packaged and released.

If you like, you can try to fix your broken config file. Open the file (backup first) in any text editor and look for entries that start with **button ...**. The number after 'button' must go from 0 to however many buttons you have, with no missing number between the numbers. The order of the numbers doesn't matter. Can be (0, 1, 2, 3, ...) or (2, 1, 3, 0, ...), but cannot be (0, 1, 4, 3) because 2 would be missing.

messy117, 2024-01-11:
super newb here, do i just copy and paste that commit coding into my pronterface.py?

rockstorm101, 2024-01-11:
> Unfortunately, this fix has not yet been packaged and released.

Maybe it is time for a 2.1.0 release? ;)

> super newb here, do i just copy and paste that commit coding into my pronterface.py?

I assume you use Windows but unsure how exactly you run Printrun. Easiest thing I think would be to try running one of the [nightly Windows builds][1].

[1]: https://github.com/kliment/Printrun/actions/runs/7481629825

DivingDuck, 2024-01-11:
@messy117,
No. Please do what I had ask for and answer the question I wrote in my reply. I will then look into it and correct your config file (if that is the problem). But first I want to see what had happen on your installation.

@neofelis2X,
Thanks for reminding this issue. You are correct. Unfortunately we still do not know what operating system @messy117 have. I did the test with version 2.0.1 as well and I can't move a custom button out of range either for left or right side because they are deactivated as soon the hit the last or first position. This is why I ask for the files.
I had as well this bug long time ago with the very first versions of 2.00RCx. I still think that this sometimes still happen as well when a user switch between 1.x and 2.x versions. But I'm not sure. I hoped with the requested data and information becoming a better idea of this problem especially as long the data are not compromised from any kind of repair attempts. There were one or two similar problems in the past where we unfortunately did not get an answer back when we ask for it.

@rockstorm101, 
Good idea. Yes we should make a new release soon. Please wait until I build a new set of translation files. I had planed to do it this week end and then ask you for a new release :)

rockstorm101, 2024-01-11:
> Please wait until I build a new set of translation files.

Understood. Take your time. No reason to rush

neofelis2X, 2024-01-14:
> Maybe it is time for a 2.1.0 release? ;)

Yes, absolutely. I won't be able to contribute to the interface for a while, because I am occupied with other things. But whats already merged should be released. :)

DivingDuck, 2024-10-18:
This issue was solved with release 2.1.0

closed 2024-10-18 by DivingDuck