kliment/Printrun #1405: Cannot download on MAC

Toxicshawk, 2024-01-17:
Neueste version 2.0.1 cannot open on Mac.  Says "apple cannot verify it for malicious code:

DivingDuck, 2024-01-17:
Please take a look on readme.md, the first page you see with most of all instructions and information: https://github.com/kliment/Printrun#windows-and-macos-pre-compiled-binaries

Toxicshawk, 2024-01-18:
Good afternoon,

I was able to get the program running... to Update the above instructions
need to hold Control then click on the app to get to the "open" screen.

However,  wondering if you have any knowledge on the anycubic kobra 2 max
machine?
Not sure if there is a way to connect my computer to the machine to
run/update calibration ????

This has been a very frustrating experience for me and any help/assistance
is greatly appreciated.

Have a wonderful day

Carey Vincent

rockstorm101, 2024-01-19:
> to Update the above instructions need to hold Control then click on the app to get to the "open" screen.

Thank you for reporting this. I'm happy to update those instructions if they are incorrect. Could any macOS user/maintainer comment on it? CC @neofelis2X, @a2k-hanlon

a2k-hanlon, 2024-01-21:
I would say the "right click to open" procedure already described in the readme as pointed out by @DivingDuck is correct, although apparently Apple has changed the wording of the warning a bit. A right click is equivalent to control-click.

@Toxicshawk So you are aware, this procedure is typical of many pieces of open source software because a developer needs to have a paid subscription to Apple's developer program, and take additional steps when releasing an app, in order to sign apps. Signing an app is what makes it "verified" and legible to open normally after first installing or updating. Not all open source projects take on this burden. It is not an issue exclusive to Printrun. I recommend keeping this in mind for other open source MacOS software you may find online.

DivingDuck, 2024-01-21:
@a2k-hanlon, 
indeed, I now recognize that we missed to add this in the description. I will add  control-click for macOS. Thanks a lot for pointing to this.
@rockstorm101,
I will add this to pull request #1406 today if this is ok for you.

DivingDuck, 2024-01-21:
@Toxicshawk ,
sorry for not answering your latest question, I was a bit busy this week and short of time.
I do not own a anycubic kobra 2 max. What exactly do you need to calibrate? 
It shouldn't be no problem to do any sort of calibration with gcode commands. In general you can enter those in the console window on the right side in the bottom input field.
Let me know what you want to do and I can look how I can help you a bit.

rockstorm101, 2024-01-21:
> I will add this to pull request #1406 today if this is ok for you.

I'm happy with that, no problem. My understanding was that the current "right click to open" procedure was correct, so no update was necessary, but I'm happy to change it to whatever makes it clearer for users if you want to reword that.

DivingDuck, 2024-06-05:
I will close this issue as #1406 is merged

closed 2024-06-05 by DivingDuck