kliment/Printrun #1408: Pause from G-Code

STS-ATS, 2024-01-28:
Hi there!

I love your work, its an amazing tool for 3d printing, thank you very much for your effort and this great job.

Ive been trying to pause the printing from the G-code, ive tried to put the pause command, pause, !p.pause(), p.pause()

But non of them have worked, and i need to automatically pause the printing for filament change.

Thanks for your suggestions.

STS-ATS, 2024-01-28:
Well, it seems like, i forgot to read a little part of the documentation. it says:

"you can either use directly or inside your G-Code. To run a host command from inside a G-Code, simply prefix it with ;@"

Ive read all the Using host commands section https://github.com/kliment/Printrun#using-host-commands
except that part of the prefix.

So, if you are having the same problem you can put in your G code, in the desired layer

;@pause      exactly how i wrote, in the console you ll see:   Print paused at: 23:06:18  its an example

or, if you do:

;@ pause    in the console you ll see:  G-Code calling host command " pause"

both makes the task of pausing the print

Hope it could be helpful for you.

Thanks to the Pronterface team for their good work.

If yow know how to show the progress bar on the ender 3 s1, would be great

thanks

rockstorm101, 2024-01-28:
> Thanks to the Pronterface team for their good work.

Thank you for your kind words. Much appreciated.

> Also, ive had some troubles trying to show the progress bar on Ender 3 S1, it makes an echo of unknown M73 command
> it would be great to control the print also from the printer but its not working.

I'm not sure what you are trying to achieve here. Print progress is shown in Pronterface in the status bar. A message along the lines of "unknown M73 command" probably means your firmware does not support such a command.

STS-ATS, 2024-01-28:
Hi, thanks for answering

Yeah, its more related to the ender firmware, but i tought maybe someone have already figured out how to show the progress bar on an ender 3 s1

But, ive noticed that some slicers like Orca Slicer wich is the one i use for 3d printing, has an option of disabling the M73, because of the compatibility (i think), thats why i have the echo of M73, the problem its not from pronterface, because it uses the M117 command, thats what confused me.

(the problem comes from the slicer xd)

But... even if the printer doesnt makes an echo of error, it doesnt works as expected.

so, maybe if you know why or someone knows, it would be pretty awesome to have a progress bar on the display
it doesnt change the performance or anything
but it would be awesome (again)

thanks