kliment/Printrun #1409: Is there a Pause Flag on RPC server?

STS-ATS, 2024-01-28:
Hi there,

Ive been working on a little code for getting a notification with the printing characteristics and sending some commands to control it remotely

But, in some Gcodes i need to know when the printing got paused to send a notification

Is there any form or a kind of "flag" that could be accesed via RPC server?
I have a couple of ideas on how to solve the problem, like reading the eta if its equal to 0, but, if its already integrated, could be cleaner solution.

Ive read the rpc.py code but, cant find something like that status report on pause

Thanks to the pronterface team for their work.

Thanks for your suggestions and answers.

DivingDuck, 2024-01-29:
One possible way is to add https://reprap.org/wiki/G-code#M118:_Echo_message_on_host before a pause. You can catch up the message then and use this as a trigger in your action chain.

STS-ATS, 2024-01-30:
Ok, got it
Thanks

i have another question, how could i know in wich exact number line of the G code, the software is sending to the printer?

DivingDuck, 2024-01-30:
> Good question (note to self to document this attribute) The `printcore.printcore.queueindex` attribute holds the current index in the queue of commands to be sent

I did never use or play myself with printcore.printcore.queueindex but I saw you mentioned this in #1397. I saw an old issue #263 where @kliment mentioned `!print self.p.queueindex` as well.

STS-ATS, 2024-03-07:
Hey there, i have an update, jejeje

Guys, the RPC server is already reporting the current line, on the ETA, its the third value, i have already been able to extract it and use it

Somehow i didn't notice it when i was working with that

Thanks, hope it would be usefull

DivingDuck, 2024-03-07:
I just run the rpc server it myself as I didn't use it in the past. I can see the following output:

On print:

```
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG_ohneTemp.gcode', 'progress': 2.4973348391931687, 'eta': [7276.893032068913, 7471.893032068913, 4732], 'temps': {'T': ['23.7', '0.0'], 'T1': ['24.0', '0.0'], 'B': ['23.4', '0.0'], 'B1': ['22.8', '0.0']}, 'z': 0.6000000238418579}
```

On Pause:
```
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG_ohneTemp.gcode', 'progress': None, 'eta': None, 'temps': {'T': ['23.7', '0.0'], 'T1': ['24.0', '0.0'], 'B': ['23.4', '0.0'], 'B1': ['22.8', '0.0']}, 'z': 0.800000011920929}
```
I can see that there are some quirks, like format of eta, z high not showing on first layer or when printing from SD do not show any information except temperature.

**Here an example where I start a real print from start to manually pausing and then aborting the print:**

Pay attention on my --> marks for understanding what happen during a print and where some wrong information occur. 

<details>

```

>>> import xmlrpc.client
>>> rpc = xmlrpc.client.ServerProxy('http://localhost:7978')
>>> print(rpc.status()) --> after start heating the bed. Target as well as actual temp and z arn't updated what is actual a bad situation...
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.005805300767355211, 'eta': [6551.908890955335, 6551.908890955335, 11], 'temps': {'T': ['24.7', '0.0'], 'T1': ['24.4', '0.0'], 'B': ['23.9', '0.0'], 'B1': ['23.2', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.005805300767355211, 'eta': [6551.908890955335, 6551.908890955335, 11], 'temps': {'T': ['24.7', '0.0'], 'T1': ['24.4', '0.0'], 'B': ['23.9', '0.0'], 'B1': ['23.2', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.007388564612997541, 'eta': [6551.820835508195, 6713.820835508195, 14], 'temps': {'T': ['25.8', '0.0'], 'T1': ['24.3', '0.0'], 'B': ['81.0', '81.0'], 'B1': ['23.2', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status()) --> after bed temp reached the target temp. Still wrong temps before and then only heater temp is shown while bed temp is at 81 °C, z isn't updated...
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.008444073843425761, 'eta': [6551.762131876768, 6719.762131876768, 16], 'temps': {'T': ['105.3', '240.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.008444073843425761, 'eta': [6551.762131876768, 6719.762131876768, 16], 'temps': {'T': ['151.3', '240.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.008444073843425761, 'eta': [6551.762131876768, 6719.762131876768, 16], 'temps': {'T': ['192.4', '240.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.008444073843425761, 'eta': [6551.762131876768, 6719.762131876768, 16], 'temps': {'T': ['219.7', '240.0']}, 'z': 1.600000023841858}
>>> print(rpc.status()) --> bed and heater reached target temp and started the print. z still not updated... 
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.031137522297632492, 'eta': [6550.500003801096, 6818.500003801096, 59], 'temps': {'T': ['245.2', '240.0'], 'T1': ['24.4', '0.0'], 'B': ['81.1', '81.0'], 'B1': ['23.4', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.10396765919717968, 'eta': [6546.44945323266, 6826.44945323266, 197], 'temps': {'T': ['245.6', '240.0'], 'T1': ['24.4', '0.0'], 'B': ['81.1', '81.0'], 'B1': ['23.4', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.2153238830073569, 'eta': [6540.256220117151, 6838.256220117151, 408], 'temps': {'T': ['238.9', '240.0'], 'T1': ['24.4', '0.0'], 'B': ['81.1', '81.0'], 'B1': ['23.4', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.29712584836554395, 'eta': [6535.706688681589, 6846.706688681589, 563], 'temps': {'T': ['240.0', '240.0'], 'T1': ['24.4', '0.0'], 'B': ['81.1', '81.0'], 'B1': ['23.5', '0.0']}, 'z': 1.600000023841858}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': 0.42748123832342916, 'eta': [6528.456790200402, 6853.456790200402, 810], 'temps': {'T': ['240.3', '240.0'], 'T1': ['24.4', '0.0'], 'B': ['81.1', '81.0'], 'B1': ['23.5', '0.0']}, 'z': 1.600000023841858}
[... cut ...] ', '0.0'], 'B1': ['23.8', '0.0']}, 'z': 0.6000000238418579}
>>> print(rpc.status())
{'filename': 'C:\\Users\\Armin\\Documents\\3D-Printer\\3D Modelle\\3DBenchy\\Armin\\3DBenchyHEVOPETG.gcode', 'progress': None, 'eta': None, 'temps': {'T': ['207.2', '0.0'], 'T1': ['24.5', '0.0'], 'B': ['75.5', '0.0'], 'B1': ['23.9', '0.0']}, 'z': 0.6000000238418579}
>>> 
```

</details>

There is real potential for improving the code of Pronterface / RPC server. Be careful with interpreting the shown information.

rockstorm101, 2024-03-09:
> @rockstorm101, we should leaf this open as I have some concern with the behavior of RPC (see in details). The shown information need to be at least correct, especially for temp values.

Agreed. We can definitely improve the code a bit. I would open a new issue with your findings and what exactly needs changing. Since this one was more like a question about pausing a print via RPC and not technically a bug report. But up to you. I'm happy either way.