kliment/Printrun #1412: AttributeError: 'NoneType' object has no attribute 'write

RiQ4, 2024-02-10:
Pronterface.exe not opening.

Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2586, in __init__
  File "printrun\pronterface.py", line 170, in __init__
  File "printrun\pronsole.py", line 786, in parse_cmdline
  File "printrun\pronterface.py", line 1154, in process_cmdline_arguments
  File "printrun\pronsole.py", line 769, in process_cmdline_arguments
  File "printrun\pronsole.py", line 664, in load_default_rc
  File "printrun\pronsole.py", line 635, in load_rc
  File "cmd.py", line 211, in onecmd
  File "printrun\pronsole.py", line 412, in default
  File "cmd.py", line 236, in default
AttributeError: 'NoneType' object has no attribute 'write'

rockstorm101, 2024-02-10:
Hi @RiQ4, assuming you use Windows. Please also state what version of Printrun are you running and where did you get it from.

DivingDuck, 2024-03-23:
@RiQ4, we can only help if there is a two way communication. My guess is that you try to run an older version of Pronterface.
Please download the actual version from here: https://github.com/kliment/Printrun/releases and give a feedback if this version runs on your system or not.

closed 2024-04-08 by DivingDuck