kliment/Printrun #1416: Multiple printer config files

snekya, 2024-02-22:
I currently run two instances of Pronterface to run two different printers.  I generally have no issues with two instances running at a time.
My goal is to create two separate config files and have to shortcuts or instances of pronterface correspond to those config .ini files.   
My question is, how do I make two different shortcuts for each printer configuration that correspond to separately named config .ini files?  
I saw issue #434 but I am little confused as to where and what I should edit to create the conditions I am seeking.  
I have opened pronterface.py in IDLE but I am still unsure of a few things to do this correctly.  
Pardon my lack of knowledge with python.  
If someone could please help walk me through this a little further than issue #434 did I would deeply appreciate it!
I know it is possible, I just don't fully understand how to do it.  I am assuming it is as simple as knowing where to place the new path and saving a separate shortcut.  

Thank you!

DivingDuck, 2024-02-23:
Hi @snekya,

below you will find the command line options for Pronterface and how to use them with Windows:
```
 usage = "Usage:\n"+\
         "  pronterface [OPTIONS] [FILE]\n\n"+\
         "Options:\n"+\
         "  -h, --help\t\t\tPrint this help message and exit\n"+\
         "  -V, --version\t\t\tPrint program's version number and exit\n"+\
         "  -v, --verbose\t\t\tIncrease verbosity\n"+\
         "  -a, --autoconnect\t\tAutomatically try to connect to printer on startup\n"+\
         "  -c, --conf, --config=CONFIG_FILE\tLoad this file on startup instead of .pronsolerc; you may chain config files, if so settings auto-save will use the last specified file\n"+\
         "  -e, --execute=COMMAND\t\tExecutes command after configuration/.pronsolerc is loaded; macros/settings from these commands are not autosaved"
```
pronterface [OPTIONS] [FILE]

in your case, if your config is in folder Desktop and the config file name is printer2:

`Pronterface.exe -c "c:\Users\snekya\Desktop\printer2.ini"`
or using the command line version 
`Pronsole.exe -c "c:\Users\snekya\Desktop\printer2.ini"`

Copy an existing config file (printrunconf.ini) and rename it to your needs. You need to add the complete path and file name and put it in quotation marks as shown above.

Same for running from source (Pronterface.py / Pronsole.py )

DivingDuck, 2024-02-23:
Maybe one additional example, you can do in Windows also things like this:

`%userprofile%\Downloads\pronterface-test\Pronterface.exe -c "%userprofile%\Downloads\printer2.ini"`

This way you don't need to write/use the complete absolute path for an active user account.

snekya, 2024-02-23:
OK!  I got it to work with the second method you showed me.  So I assume I can save that as a shortcut path and use that?

snekya, 2024-02-23:
I have the correct path saved and I can type in that in explorer and it works with the config correctly.   I am trying to now figure out how to save it as a shortcut of some kind.

snekya, 2024-02-23:
I really appreciate your help!  I am now running two different instances with separate configs and they are saving and loading correctly.  If you have any tips on how to make these shortcuts I appreciate it.  But, problem solved!  thank you again.

snekya, 2024-02-23:
Saved as a batch file. Great success.  I learn something new every day.  Thank you kindly!

closed 2024-02-23 by snekya

DivingDuck, 2024-02-23:
Glad that you find a solution that works for you. It was too late for me (around 3 o'clock in the morning in my country).  :)

Regarding shortcuts, You can create a desktop shortcut and then modify it in the way you need it. Here is an example how to run the command line version Pronsole.exe in a windows terminal. The screenshot is in German but I'm sure you find the same view in your language as well.
This is documented here: https://github.com/kliment/Printrun?tab=readme-ov-file#windows (look for ...Running Pronsole works best by running it in WindowsTerminal. ) You may want to install Windows Terminal if you don't have it already installed.

First create a desktop shortcut  by [mouse right click] on a file like pronsole.exe. Then edit the Desktop shortcut by right clicking and select [Properties], choose the tab [Shortcut] and modify the the field [Target] and [Start in:]

Target (Ziel):
`%LocalAppData%\Microsoft\WindowsApps\wt.exe --title Pronsole %USERPROFILE%\Downloads\pronterface-test\Pronsole.exe`
Start in (Ausführen in:):
`%USERPROFILE%\Downloads\pronterface-test`

In this example I call windows Terminal with `%LocalAppData%\Microsoft\WindowsApps\wt.exe`, name the shortcut as Pronsole with `--title Pronsole` and run pronsole with `%USERPROFILE%\Downloads\pronterface-test\Pronsole.exe`. And when you extend this with your individual config file, like `-c "%userprofile%\Downloads\printer2.ini"` you will have what you need. In the same way you can start an executable binary file or Pronterface from code.
A lot of possible way's running Pronterface or Pronsole.

[image]

snekya, 2024-02-23:
Thank you so much for your assistance and thoughtful reply.   I will try this as well.  The batch file worked great but I will do this as well to understand this better.  I thank you once again for the thorough explanations and for taking the time to assist me with my issue.