kliment/Printrun #1417: faild to execute script pronterface

surajkiritoo, 2024-03-11:
[image]

rockstorm101, 2024-03-11:
Hi @surajkiritoo, assuming you use Windows. Please also state what version of Printrun are you running and where did you get it from.

DivingDuck, 2024-03-11:
... and what version of Windows as well.

DivingDuck, 2024-03-23:
Isn't this frustrating? A user did the first issue report ever and then we get no answer back of a easy to answer question.
The actual version 2.0.1 should run on Windows 10 fine. Please consider to download it from here: https://github.com/kliment/Printrun/releases.

Guess, we can close this issue because of missing feedback.

rockstorm101, 2024-03-24:
> Isn't this frustrating? A user did the first issue report ever and then we get no answer back of a easy to answer question. 

It is frustrating indeed. Sadly, a lot of people are not used to reporting issues (in general, not just here or on GitHub) and expect to get help with the usual plain "it doesn't work" message. I've seen that hundreds of times. Reporting is a skill that maybe should be taught at school or something.

> Guess, we can close this issue because of missing feedback.

Agreed, closing.

closed 2024-03-24 by rockstorm101