kliment/Printrun #1419: what's the game?

Akcamay, 2024-03-23:
I'm trying to download Ponterface but instead I get a run around that does not have the download file.  What's the game?  is it free or are you in this to frustrate people?

DivingDuck, 2024-03-23:
I don't get what you try to express. There is absolutely no "game" here. The download is available and is works fine.

Akcamay, 2024-03-23:
Well, I'm missing something.  I got a download but it was not the program.
[https://res.cdn.office.net/assets/mail/file-icon/png/zip_16x16.png]pronterface-app.zip<https://1drv.ms/u/s!AsIOnteCUiBGxFj1obYUWu7899vf> Pronterface comes as a file not an app or program.
Not sure what happened when I extracted from this zip file but there is not a program that I can find in it.

Rev. Curtis A. May
Serving St. John Lutheran Church
Church Web Site http://www.stjohns-church.com/home.html
***@***.***
Church (574) 223-6898
Pastor's Cell (547) 835-8355

kliment, 2024-03-23:
Not sure what you got but here are the installation instructions for various operating systems: 

https://github.com/kliment/Printrun?tab=readme-ov-file#getting-printrun

The current releases are at: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.1

How you install/run after download depends on your operating system - can you specify what operating system you are using, what you downloaded, and what you tried to do to get it running? Unfortunately the link in your email does not work for me.

Akcamay, 2024-03-23:
Thank you, I will try them out later.  Got to go to work now.

Rev. Curtis A. May
Serving St. John Lutheran Church
Church Web Site http://www.stjohns-church.com/home.html
***@***.***
Church (574) 223-6898
Pastor's Cell (547) 835-8355

https://github.com/kliment/Printrun?tab=readme-ov-file#getting-printrun

kliment, 2024-03-23:
If you're on a mac, there's a second zip inside the zip you downloaded, which contains the app (which you can drag to your applications folder to install). The zip-inside-a-zip is a quirk of how github packages releases - you can point it to a single file (and we do) but it still insists on putting it in a zip of its own. Sorry about the confusion related to that.

Akcamay, 2024-03-23:
Thank you, but I don't use a Mac.  one should never play with fruit, just eat it.  Macintosh is a type of apple.  LOL

Rev. Curtis A. May
Serving St. John Lutheran Church
Church Web Site http://www.stjohns-church.com/home.html
***@***.***
Church (574) 223-6898
Pastor's Cell (547) 835-8355

rockstorm101, 2024-03-24:
> is it free or are you in this to frustrate people?

Definitely not the friendliest approach to ask for help. Let me remind this is an open source project, so we are all volunteers here. Please mind your manners Reverend.

> Thank you, but I don't use a Mac. one should never play with fruit, just eat it. Macintosh is a type of apple. LOL

We still don't know what your operating system is or how/where you got your download. We can't help you if you don't provide a bit of context to your issue.

Akcamay, 2024-03-25:
My apologies, I'm not trying to insult anyone, which seems to easy to do in this age.  I'm frustrated trying to find solutions that seem to be hard to achieve.  My operating system is Windows 10 on a Power Spec computer from Micro Center.  I was trying to find a program that would allow me to send G-code to my Ender 3 Max Neo.

As a pastor I do not consider myself to be perfect, nor without sin as all people are sinners and non are perfect.  I see much on the computer that frustrates me as one cannot always see the scam through the forest.  to much advertising that tries to force you to use their product.  Great that you volunteer, but please have some empathy for everyone.

I fixed my problem with my Ender 3 Max Neo.  and I believe I can use NotePad++ for any G-code additions.  Thank you for putting me in my place, and truly may God bless you always.

Rev. Curtis A. May
Serving St. John Lutheran Church
Church Web Site http://www.stjohns-church.com/home.html
***@***.***
Church (574) 223-6898
Pastor's Cell (547) 835-8355

BrunoTheGreat, 2024-03-25:
Thank you Kliment,
I have to say, I was kind of in the same predicament as Akcamay, extremely rare user of GitHub, and often lost on how to get things running unless its a simple download and run. This time, with my office computer, things were even more complicated by all the permissions needed ( password needed 8 times to install python) and the fact that I needed that github install program, python, and c++. It's been a month now since I have been wanting to bring my 3d printer back to life in order to print a part to reattach the hinge on my personal laptop. Chicken or egg, needed laptop to print its own part to be able to turn it on. Hah. Thanks again for instructions.

BrunoTheGreat, 2024-03-25:
Actually, I can't seem to download the windows version due to network issues. the macos version downloads fine.

DivingDuck, 2024-04-08:
Is there anything we can help with or should we close the issue?
Best regards,
DD

Akcamay, 2024-04-08:
Thank you very much for asking.  I am a happy camper.  You can close this issue.  And, thank you for all the help you gave, even and especially the spiritual help that set me back on the right path.

Rev. Curtis A. May
Serving St. John Lutheran Church
Church Web Site http://www.stjohns-church.com/home.html
***@***.***
Church (574) 223-6898
Pastor's Cell (547) 835-8355

closed 2024-04-11 by rockstorm101