kliment/Printrun #1421: Problems if try to print a file, just after print the previous one

Escrich, 2024-04-08:
Dear all, first of all congratulations by create Pronterface, I'm using it from years ago, and I love it.

This time I have an old issue, happening at least on versions from last three years I bet, the problem arises when you print a common gcode, in my case always generated by Prusa slicer, and just, after finish the print, made from Pronterface, (I use Pronterface to avoid using the SD card), then, just if you finish to print a part, you simply load another file on Pronterface, and click on print button, instead to print, you only get a stream of text showing you error after error, in the last three years, my procedure is so simple, I simply click on disconnect, then in connect, again, ant this manoeuvre generates a reset on my Prusa original Mk3s printer, then you can print the same gcode that was "generating" errors, without any problem, but always you need to make a reset in between files, now I'm using latest Pronterface versión, but I can get same behaviour in both versions you could see in the attached pictures:

[image]

[image]

rockstorm101, 2024-04-09:
> Dear all, first of all congratulations by create Pronterface, I'm using it from years ago, and I love it.

Glad you like it, thanks a lot.

> [...] just if you finish to print a part, you simply load another file on Pronterface, and click on print button, instead to print, you only get a stream of text showing you error after error [...]

Could you please paste some of those errors? Also, what's your OS?

Escrich, 2024-04-15:
Sorry, for the late answer, I found your message just now, OS is Windows 10, I was sending this gcode now:
[20240415 Prusa MK3s Marca de centro_0.2mm_PETG_MK3S_1m.gcode.zip](https://github.com/kliment/Printrun/files/14983146/20240415.Prusa.MK3s.Marca.de.centro_0.2mm_PETG_MK3S_1m.gcode.zip)

Normal status printing the first file after reset or connect:
[image]

Status once printed:
[image]

Behaviour once printed the first file, and trying to print the second once finished the first, simply after click on print button again:
[image]

The messages on Pronterface are scrolling, showing same kind of error, at end nothing happens, the heating don't start, and the printer still shows the text "Finished print!"

DivingDuck, 2024-04-15:
Hi,
can you please check this version of Pronterface whether the problem still exist:
https://github.com/kliment/Printrun/actions/runs/8681026406
This is the latest development version.
Best regards,
DD

DivingDuck, 2024-04-17:
I can confirm this error. During my test I was able to see this as well for Smoothieware and an old Marlin version for both printing direct with Printrun and via SD card. It looks like Pronterface do not clear the buffer when it finish or cancel a print.

I haven't a solution for now. @rockstorm101, any idea's?

rockstorm101, 2024-04-21:
Hi, sorry for the delayed response, I'm abroad these months so I can't really test for myself with a physical printer. I'm not able to reproduce this at the moment. Those errors sound familiar though, see #1341. It's an error sent from the firmware, hence why it can't be reproduced with a mock-up printer I guess.

Could you go in verbose mode and check that the "M110 N-1" command is being sent and received ok when the print starts and finishes?

DivingDuck, 2024-04-23:
Hi,
no problem. This was the issue I was searching for. I will look on this as soon I find a little bit time this week.

DivingDuck, 2024-08-31:
Hi @Escrich,
sorry that it takes so long for a follow up.
Today I had a little bit time for digging a bit deeper to this issue and I think I identified the problem.

The problem started with the implementation of a workaround for Prusa printers and their firmware update to version 3.12.2 where they changed a behavior that was common for all firmware manufacturer in the past. I don't want to repeat the story here, you can find the discussion here: #1341 and #1378.

Anyway, with this change we introduce a compatibility issue for other firmware manufacturer than Prusa and user of older firmware, that do not support the change. I revert this change for a test with two printers that runs Marlin 1.x and Smoothieware 1.x firmware and can then repeat prints without any problems.

I have one last question: Had your printer an older firmware version from before 3.12.x when you report this issue?

@rockstorm101,
I changed in printcore.py line 419 and 678 the actual behavior ..."M110 N-1...  back to the old notation ..."N-1 M110"... for my test.
It looks like we now need to break with what we had tried to avoid all the time for not dealing with special behaviors for printer manufacturer and firmware versions. I must admit I'm not really a fan of this as we can not say or test what devices (in combination of board manufacturer and their firmware versions) in the wild are now affected and cannot repeat print jobs w/o resetting and reconnecting the printer first for a next print.
A minimal workaround could be that we add a additional global setting via setup where a user can set/change the behavior manually via the config file and set the old behavior back as default 😇. I guess we need to add this in the GUI via option menu as well if we don't want to push users to modify the config file manually.
Let me know what you think about this.

rockstorm101, 2024-09-08:
> I revert this change for a test with two printers that runs Marlin 1.x and Smoothieware 1.x firmware and can then repeat prints without any problems.

That's weird, at least for Marlin. According to (some) documentation [[1]] [[2]] the command `M110 N-1` should work Ok on all firmwares unless I'm missing something. Marlin accepts the "backwards" option, say `N-1 M110` but others don't, hence why the change in #1378.

On configurations which trigger errors, if you look at the logs:
 * Can you see the `M110 N-1` command being sent at the begging of the print?
 * Is it sent at the end of the print?
 * Is it being sent "as is" plain `M110 N-1`? or with a checksum or something else?
 * Is the printer responding `ok` to this command? or is not being understood?
 * When you hit print again, is it being sent?
 * Errors begin right after the M110 command is sent? On first line of next print?

Ideas that come to mind rather than trying to split the behavior based on firmware:

 * Set the line number at the end of a print as if it was a "normal" print line:
    ```diff
    diff --git a/printrun/printcore.py b/printrun/printcore.py
    index f0f3da8..044da4a 100644
    --- a/printrun/printcore.py
    +++ b/printrun/printcore.py
    @@ -673,9 +673,9 @@ class printcore():
                 self.printing = False
                 self.clear = True
                 if not self.paused:
    +                self._send("M110 N-1", self.lineno, True)
                     self.queueindex = 0
                     self.lineno = 0
    -                self._send("M110 N-1")
     
         def _send(self, command, lineno = 0, calcchecksum = False):
             # Only add checksums if over serial (tcp does the flow control itself)
    ```

 * This might be silly but... If the problem is in fact that old firmware do not understand the "forward" notation of M110 command, could we try to use both notations one after the other? I.e.:
    ```
            if not self.paused:
                self.queueindex = 0
                self.lineno = 0
                self._send("M110", -1, True)
                self._send("M110 N-1")
    ```

As you see I would avoid customizing the code for particular firmwares as much as possible, but if we get to that, I agree that a GUI toggle/checkbox does sound like the way to go.

[1]: https://marlinfw.org/docs/gcode/M110.html
[2]: https://reprap.org/wiki/G-code#M110:_Set_Current_Line_Number

DivingDuck, 2024-09-09:
Hi @rockstorm101,

yea, I saw this as well. I don't know if I remember correct, whether that was valid for older versions of Marlin too. I had vaguely in mind that there was a discussion long time ago for this topic like e.g. here: https://github.com/Smoothieware/Smoothieware/issues/1228. I found several discussions about this topic and the attempt to change the behavior, but in the past it was common understanding not to change it because of not breaking older implementations. 
 
> * Can you see the `M110 N-1` command being sent at the begging of the print?
> * Is it sent at the end of the print?

yes it will be send for both

Actual second print attempt (with M110 N-1), not working:

<details><summary>Show log...</summary>

```
SENT: M110 N-1
RECV: ok
Print started at: 21:07:17
SENT: N0 G28*19
RECV: rs N34
SENT: N1 G0 Z10 S0*16
RECV: rs N34
SENT: N2 G90*18
RECV: rs N34
SENT: N3 G21*25
RECV: rs N34
SENT: N4 G1 F1000*75
RECV: rs N34
SENT: N5 G1 X122.4164 Y129.3071*37
RECV: rs N34
SENT: N6 G4 P0*107
RECV: rs N34
SENT: N7 G0 Z0 S0*39
RECV: rs N34
SENT: N8 G4 P0*101
RECV: rs N34
SENT: N9 G1 F1000.000000*104
RECV: rs N34
SENT: N10 G1 X173.5241 Y129.3071*16
RECV: rs N34
SENT: N11 G1 X173.5241 Y64.5357*40
RECV: rs N34
SENT: N12 G1 X169.0267 Y64.5357*33
RECV: rs N34
SENT: N13 G1 X169.0267 Y68.0069*39
RECV: rs N34
SENT: N14 G1 X126.9139 Y68.0069*42
RECV: rs N34
SENT: N15 G1 X126.9139 Y64.4663*47
RECV: rs N34
SENT: N16 G1 X122.4164 Y64.4663*45
RECV: rs N34
SENT: N17 G1 X122.4164 Y129.3071*22
RECV: rs N34
SENT: N18 G4 P0*84
RECV: rs N34
SENT: N19 G0 Z10 S0*41
RECV: rs N34
SENT: N20 G1 F1000*125
RECV: rs N34
SENT: N21 G1 X108.5735 Y112.7845*27
RECV: rs N34
SENT: N22 G4 P0*93
RECV: rs N34
SENT: N23 G0 Z0 S0*17
RECV: rs N34
SENT: N24 G4 P0*91
RECV: rs N34
SENT: N25 G1 F1000.000000*86
RECV: rs N34
SENT: N26 G1 X90.642 Y112.7845*24
[... 371 lines cut ...]
SENT: N30 G0 Z10*97
RECV: ok
SENT: N31 G0 X5*86
RECV: ok
SENT: M105
RECV: ok T:22.8 /0.0 @0 T1:21.9 /0.0 @0 B:22.0 /0.0 @0 B1:21.5 /0.0 @0
SENT: N32 G0 Y5*84
RECV: ok
SENT: M105
RECV: ok T:22.9 /0.0 @0 T1:21.9 /0.0 @0 B:22.0 /0.0 @0 B1:21.4 /0.0 @0
SENT: N33 M84*47
RECV: ok
SENT: M110 N-1
SENT: N-1 M110
RECV: ok
RECV: ok
Print ended at: 21:46:59 and took 0:00:47
SENT: M117 Finished Print
RECV: ok
SENT: M105
RECV: ok T:22.9 /0.0 @0 T1:21.9 /0.0 @0 B:22.0 /0.0 @0 B1:21.5 /0.0 @0
```
</details>

I think we should implement this as that way it will not break older firmwares. I like to test this in addition with an old Marlin firmware implementation. That printer will come back tomorrow or the day after. I will push an update for this after the test if you are ok with the change.

DivingDuck, 2024-09-12:
Hi,

I need a tester with a Prusa printer and firmware >=3.12.x. Other printer firmwares for testing are as well welcome.

Long story short, I found in my test an additional firmware from Marlin v1.0.0 that behave different to newer Marlin versions.So we have now 3 variants:

- Prusa firmware >3.12.x wants `M110 N-1` for resetting the gcode line number in the printer
- Smoothieware wants `N-1 M110`
- Old Marlin firmware 1.0.0 ... x wants `N-1 M110 + line checksum`

My solution for this is adding back the old behavior and combine this with the actual implemented Prusa modification:
`self._send("M110 N-1", -1, True)`
This ends in `N-1 M110 N-1*125` and both, the old firmware from Marlin and for Smoothieware are happy with the gcode. **Now I need one who can test this solution for the Pusa firmware.**

**Link for Pronterface test version in my repository:**
Windows: **https://github.com/DivingDuck/Printrun/actions/runs/10839165094**
MacOS: **https://github.com/DivingDuck/Printrun/actions/runs/10839165224**

In addition here is a little gcode file for a quick test (area needed from homing position: x78 y70 z10 mm) that will run only a half minute and prints above the print bed (z=5 mm) w/o heating and extrusion:
**https://github.com/DivingDuck/Printrun/blob/issue-1421/testfiles/quicktest.gcode**

Best regards,
DD

jumaka, 2024-10-01:
> Hi,
>
> I need a tester with a Prusa printer and firmware >=3.12.x. Other printer firmwares for testing are as well welcome.
>
> Long story short, I found in my test an additional firmware from Marlin [...]

I agree with @DivingDuck  I have an old Marlin based Velleman K8200, and the change has caused issues, particularly the lack of the checksum flag being set to True which means the M110 command now errors and gets stuck in an infinite loop unless the printer has been fully powered down between prints. `self._send("M110 N-1", -1, True)` works perfectly for my printer.

rockstorm101, 2024-10-05:
> So we have now 3 variants:
> 
>  1. Prusa firmware >3.12.x wants `M110 N-1` for resetting the gcode line number in the printer
>  2. Smoothieware wants `N-1 M110`
>  3. Old Marlin firmware 1.0.0 ... x wants `N-1 M110 + line checksum`

Hi, sorry I was AFK past weeks, let me see if I get where we stand on this. Do I understand correctly that the solution below works well for variants 2 and 3 but we still need to test if it works on variant 1? 

```diff
diff --git a/printrun/printcore.py b/printrun/printcore.py
index f0f3da8..042c94e 100644
--- a/printrun/printcore.py
+++ b/printrun/printcore.py
@@ -416,7 +416,7 @@ class printcore():
             return True
 
         self.clear = False
-        self._send("M110 N-1")
+        self._send("M110 N-1", -1, True)
 
         resuming = (startindex != 0)
         self.print_thread = threading.Thread(target = self._print,
@@ -675,7 +675,7 @@ class printcore():
             if not self.paused:
                 self.queueindex = 0
                 self.lineno = 0
-                self._send("M110 N-1")
+                self._send("M110 N-1", -1, True)
 
     def _send(self, command, lineno = 0, calcchecksum = False):
         # Only add checksums if over serial (tcp does the flow control itself)
```

DivingDuck, 2024-10-05:
Hi @rockstorm101,
welcome back and you are well 🙂
Yes, all what is missing so fare is a test with a Prusa printer and firmware >3.12.x which I don't own.

A test for Klipper and Reprap would be nice too. I don't expect problems there but a test would be a benefit. That's why I build and link the test version in my repository before pushing it here.

jumaka, 2024-10-05:
All

The issue for the Marlin is the lack of the checksum. Isn't that covered by
the code in the _send function. Why is this only call that has been
introduced that does not pass True and let _send sort this out. Then we are
back to one variant.

Justin Saunders

rockstorm101, 2024-10-06:
Dear @dMAC95, long story short, the change we introduced in #1378 did not play nice with older firmwares and has brought up issues like this one. Error occurs when trying to print one print right after another. So we would need someone who uses Prusa's firmware to test [this change][1]. Could you give us a hand here? Thanks in advance

[1]: https://github.com/kliment/Printrun/issues/1421#issuecomment-2395002031

rockstorm101, 2024-10-06:
> Why is this only call that has been introduced that does not pass True and let _send sort this out.

The issue was some firmware did not accept `N-1 M110` instead of `M110 N-1`. Adding checksums also adds the `N##` bit before the command and therefore it was considered problematic and the checksum was removed.

dMAC95, 2024-10-06:
@rockstorm101 I'll see what I can do, might take a day or two.

dMAC95, 2024-10-06:
Relates to #1341

DivingDuck, 2024-10-12:
I pushed PR #1454 as fix for this issue.

For testing you can use this artifacts:
Windows: https://github.com/kliment/Printrun/actions/runs/11304666766
macOS: https://github.com/kliment/Printrun/actions/runs/11304666779
Python sdist and wheels: https://github.com/kliment/Printrun/actions/runs/11304666763

DivingDuck, 2024-10-13:
@dMAC95,
had you a chance to test the changes I made? I really would appreciate a test with a Prusa machine before we merge the PR. 

You can use one of the Pronterface artifacts above and "print" this file for a quick test: 
https://github.com/DivingDuck/Printrun/blob/issue-1421/testfiles/quicktest.gcode.
The test should took only a minute or two for repeating the test print in the air w/o consuming any filament.

Best regards,
DD

dMAC95, 2024-10-14:
@DivingDuck apologies, work has been busy. I'll do it tomorrow and report back

dMAC95, 2024-10-15:
Everything passed on my end using the provided test file. I attached the printer output as presented in the GUI. Didn't see any concerning / related output from the terminal. Apologies again for the delay. I meant to last week but it slipped my mind & was busy.
[print_log.txt](https://github.com/user-attachments/files/17378105/print_log.txt)

DivingDuck, 2024-10-15:
Thank you very much for taking the time, @dMAC95. I do very much appreciate this.
One open question (just to be sure on this part as well), did you repeat the print after finishing the first print as well?

dMAC95, 2024-10-15:
Figured out all the testing I've done was through the pip package printrun instead of the downloaded repository printrun, so none of my tests were affecting anything (used requirements.txt command for package install)

_Fixed test environment, uninstalled printrun package and working from printrun pull_ #1454

Recreated original error with firmware 3.12.2: (Changed M110 line strings to "N-1 M110")
- Print 1: initial log line of "Error:Line Number is not Last Line Number+1, Last Line: -1", successful print
- Print 2: no print action, runs through entire gcode file with output of "Line Number" error for each gcode line

Printing with restored printrun fix on 3.12.2:
- Print 1: worked
- Print 2: worked
- Print 3: worked

Test with most recent Prusa firmware version 3.14.0:
- Print 1: worked
- Print 2: worked
- Print 3: worked

Only modified self.loud and logging against original pull

_Testing conducted on Prusa _

[printcore.log](https://github.com/user-attachments/files/17380189/printcore.log)
Logs got cutoff, unsure why

dMAC95, 2024-10-15:
Everything checks out on my end, if you need more Prusa testing feel free to give me a shout 👍

dMAC95, 2024-10-15:
I have an MK4 and XL for testing as well.

dMAC95, 2024-10-15:
> This is very kind. I do not expect a problem, but a quick test on those printer would be nice too 😇

Sorry, I do have to get to work now, but if you need the testing for those printers I can find time again this weekend

DivingDuck, 2024-10-15:
No problem, this is life. Thank you for taking the time.

commit 991fa184 (2024-10-15): Fix regression on M110 command (#1454)

 * Add test file for repeated prints
 * Add old behavior of M110 back and combine it with the actual behavior of Prusa firmware >3.12.x. Actual Prusa firmware >3.12.x needs "M110 N-1", Smoothieware firmware needs "N-1 M110" and old Marlin firmware 1.0.0 needs in addition line checksum. `self._send("M110 N-1", -1, True)` results in sending "N-1 M110 N-1*125" and does not harm Smoothieware and old Marlin firmware.

Fixes #1421
closed 2024-10-15 by rockstorm101