kliment/Printrun #1429: Translation files for macOS

DivingDuck, 2024-05-29:
@rockstorm101,
I recognized today that I didn't bundle the translation files to the macOS bundle as I did it for Windows builds. Makes it sense to add the locale folder to the macOS bundle for a future release?

rockstorm101, 2024-05-29:
Sure, I don't see why not. We might need to add instructions to the README or somewhere else to tell users how to use those files though. Which makes me think we could simply point to the files on GitHub then instead of distributing them. I'm unsure on what's best for users honestly so I'm easy here, whatever you think is best.

DivingDuck, 2024-05-30:
I was thinking the same but I guess most users do not to recognize that we support translations (or most possible, our users have English skills). That's why I add them to the windows distribution.
I will add a chapter to the README as well. It is may be a good chance to find other users helping for updating the translations as most of the existing are in a really bad shape.

rockstorm101, 2024-05-31:
> most users do not to recognize that we support translations

Sadly, that's very likely the current situation.

> I will add a chapter to the README as well. It is may be a good chance to find other users helping for updating the translations as most of the existing are in a really bad shape.

Totally agree. It'd be a great addition to the documentation 👌

DivingDuck, 2025-02-21:
This is now implemented with PR #1465 .

closed 2025-02-21 by DivingDuck