kliment/Printrun #1433: Error of 2D viewer when loading gcode files with G2/G3 arcs

neofelis2X, 2024-06-17:
Hi, I just noticed that when I use the 2D viewer and try to load a gcode file with G2 and G3 movements (arcs), then I get the following error message. And the visualisation fails.
```
Traceback (most recent call last):
  File "printrun\gviz.py", line 432, in paint
  File "printrun\gviz.py", line 384, in repaint_everything
  File "printrun\gviz.py", line 351, in _drawarcs
TypeError: DC.DrawArc(): arguments did not match any overloaded call:
  overload 1: argument 1 has unexpected type 'numpy.float64'
  overload 2: argument 1 has unexpected type 'numpy.float64'
```
As an example you can try [this file from printables](https://www.printables.com/de/model/902359-snap-buckle/files). But it seems to happen with all files that contain G2/G3 moves.

It is the same error on windows and macOS. It occurs from version 2.0.0 but NOT in version 2.0.0rc8.
Probably pretty easy to fix.

DivingDuck, 2024-06-17:
Good catch. I had test this with 3D view as I had wrote a test macro for testing my firmware but never tried to test it in 2D view...

Guess I know where the problem is. This came with Python 10 introduced stricter type verification. We had a lot of troubles with this in the past. It seems to be one of the so fare not identified parts of the code in this module.

gviz.py, line 334+:
```
    def _arc_scaler(self, x):
        return (self.scale[0] * x[0],
                self.scale[1] * x[1],
                self.scale[0] * x[2],
                self.scale[1] * x[3],
                self.scale[0] * x[4],
                self.scale[1] * x[5],)
```

I will change this.

Macro ArcModeG2G3:
```
G2 X10 Y20 I30 J40 F8000
G3 X10 Y20 I30 J40 F8000
G2 X10 Y20 I30 J40 F8000
G3 X10 Y20 I30 J40 F8000
G2 X10 Y20 I30 J40 F8000
G3 X10 Y20 I30 J40 F8000
M84
```

DivingDuck, 2024-06-17:
Ready for test: https://github.com/kliment/Printrun/actions/runs/9549004098

neofelis2X, 2024-06-17:
That was superfast! And yes, works perfectly.

commit 17729de3 (2024-06-17): Issue #1433. Correction of type error for G2/G3 arc moves (#1434)

Fixes #1433
closed 2024-06-17 by rockstorm101