kliment/Printrun #1435: Bacup to earlier layer after fail

drmcnelson, 2024-06-17:
A common scenario, fiber stops feeding and the print job continues for a while before it is discovered.

We can pause the print job and correct the feed problem.  But,  how do we tell pronterface to go back a few layers and resume from there?

Thank you

DivingDuck, 2024-06-18:
This isn't a functionality that is build-in for Pronterface. Maybe you can use a macro functionality for manipulating the print que but I didn't ever myself. 

In addition there are some problems involved that need to be solved. E.g, you don't know what is in the buffer que of the controller waiting for processing at the moment you stop the print. This part of gcode is for Pronterface already printed. You also need to adjust the extrusion values as they wont match any longer with the then new start point and will extruding / retraction to the value that is in the gcode. That point is maybe the easiest one to overcome with relative extrusion. Other things are acceleration and speed values that you need to recover. Homing all axis is needed so that you have real position values as most modern boards will turn off motors after some time. Or you move the print head around for e.g. cleaning or what ever else you need to do. And so on. There is quite some work involved. Given this and the fact that there are as well differences with all the firmware implementations makes it not to a simple "start from a defined layer number" implementation.

I did this in the past sometimes manually by guesstimating the layer number where the problem starts, then modify the gcode manually and use the new file for printing from the guesstimated point to finish the print (or the point where the next problem occur when it went really bad). But this is nothing you want to do on a regular base as the result isn't invisible. It's a stopgap if you need to do it, e.g. if you have a project with costly materials or with long printing time like a couple of days and huge amount of filament involved.

There was a attempt to implement such a function in the past but the development stop somewhere as there was not a lot of interest from users.

You can find here some more information: 
https://github.com/kliment/Printrun/issues/383
https://github.com/kliment/Printrun/pull/1197

drmcnelson, 2024-06-18:
@DivingDuck 

I think starting it as a new job at the specfied layer number and height. would be fine.

After homing, it needs to lift to a height that is significantly higher and the move to the starting position in x,y and then lower itself.

The imperfection is perfectly okay in the situations were I would be looking to use it.

So in another words, from what you said, what we need is a facility to create the gcode file that does that, probably with the old gcode file and a height as its inputs. 

It seems straightforward, read the old file, output the start of the job to a new file, read your way through it to the specified point, position there and resume outputting.

DivingDuck, 2024-06-19:
> So in another words, from what you said, what we need is a facility to create the gcode file that does that, probably with the old gcode file and a height as its inputs.

That is correct.
The first link of my previous answer shows a batch script from a user. This is maybe a good starting point for you. I would suggest making a process list with steps you need to follow for your printer / firmware configuration. This help to take care that you do not miss a step (this is what I do). 

It is also possible to put some functionality in a macro and a custom button as actions.