kliment/Printrun #1436: App crashes when open on multiple users (macos)

phamleduy04, 2024-06-19:
When running the Pronterface application on macOS, it crashes if the user switches to a different user account without first quitting the app. The issue occurs when attempting to open Pronterface under the second user account while it's still running for the initial user.

Steps to Reproduce:
- Launch Pronterface on a macOS system.
- Without quitting Pronterface, switch to a different user account on the same macOS system.
- Attempt to open Pronterface under the second user account.
- Observe that the application crashes.

DivingDuck, 2024-06-19:
Pronterface isn't made for this use case, it is not a multi user server. When you switch to an other user account you will loose the control of the usb connection and this will the cause to the crash. 

What is your use case that you want to do this?

phamleduy04, 2024-06-19:
my Printrun version is 2.1.0

My use case is I have a dedicated machine for 3d printing, and people will log in to their account to print. There is a case in which the user sends the print command, and locks the screen so the print can continue. However after it is done, there is a manual task that requires the user to send a print command to close/log off the machine for the next person to keep printing.  I want the next user can open the app and start print without have to wait the previous user to close the app.

rockstorm101, 2024-06-19:
As already explained, Printrun is not intended to be used in that way, it is just a desktop app. Two options come to mind for what you try to achieve:
 * Create a user for printing. Every person that needs printing will know the credentials for this account and therefore anyone can interact with the printer using this user account instead of their personal one.
 * As hinted by @DivingDuck, use a host that runs a server in the background like Octoprint.

closed 2024-07-04 by phamleduy04