kliment/Printrun #1440: problem when changing speed slider

eddie120678, 2024-07-22:
I got the following error when trying to slow down the print speed during a print. How do I go about correcting this error so that the function work as intended.

root.speed_slider.SetValue(value)
TypeError: Slider.SetValue(): argument 1 has unexpected type 'float'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_pronterface.1000.crash'

DivingDuck, 2024-07-22:
Hi,
what version of Pronterface and OS do you use?

eddie120678, 2024-07-22:
2.0.0rc7

DivingDuck, 2024-07-22:
This is a quite old version and known for some bugs. Guess, you are using a Linux operating system? Please update to the actual version 2.1.0. You may need to run it from source in case there is no actual version available for your distribution.

eddie120678, 2024-07-22:
Thanks, I'll check it out. This was the version I found through the website

eddie120678, 2024-07-22:
I am trying to build from source and believe I have installed the wheels
needed for my platform but still having an issue with install

See attached images

DivingDuck, 2024-07-23:
Hi,
I can't see any images. Follow the instructions of the main GitHub project page and use a virtual environment for installing the needed dependencies.

DivingDuck, 2024-07-27:
Hi @eddie120678,
Was you able to run Pronterface from source and did it solve your problem?

DivingDuck, 2024-07-30:
@eddie120678,
I will close this issue as this problem should not happen with the actual version.
Best regards,
DD

closed 2024-07-30 by DivingDuck