kliment/Printrun #1441: constant ok messages

Tigerwraith, 2024-08-04:
Since I updated from stock Creality firmware from https://github.com/mriscoc/Ender3V2S1 for my Ender. The command window has a constant ok messages populating while the printer is printing. Is there a way to turn these off or filter them out?
[image]

DivingDuck, 2024-08-22:
@Tigerwraith 
sorry for the late response. What version of Pronterface do you run? 
There is no filter function available. This do lot look like you have enable debug communications. This is maybe a setup in your firmware or printer setup you can change.

DivingDuck, 2024-08-22:
Thanks for the Info. 
It is maybe a good idea to ask the people who build this firmware why the behavior change and how you can prevent this continuous logging/debugging information.

Here is something you might like to check: https://marlinfw.org/docs/gcode/M111.html

DivingDuck, 2024-09-23:
@Tigerwraith,
I guess we can close this issue. There is actual no plan for filtering communication messages in addition to what is implemented with debug communication. I'm not sure if this really makes any sense for Pronterface for a future update as this kind of logging only introduce a hell lot of traffic in communication. In addition it looks for me  redundant to the implemented send&receive check via checksum for each line send to the printer.

@rockstorm101 @kliment,
any though from your point of view?

Best regards,
DD

rockstorm101, 2024-10-05:
> @rockstorm101 @kliment,
> any though from your point of view?

Well, here are my two cents.
 * After a quick look to OP's image seems like it is sending a command in the form `P0 B15` which I could not find any documentation for. So it seems the slicer is injecting that command onto the G-code file and the printer firmware is (somehow) accepting it.
 * Does it send the `ok ...` message for every single command? Maybe debug communications was turned on by mistake? And it is as simple as turning it off
 * On the other hand, if the `ok ...` message is being displayed only for commands like `P0 B15` then we (Pronterface) have an issue because we should not be displaying those "ok" messages when not in debug mode. In which case I would not close the issue yet since something is going on on our side.

All that being said, I guess we would need:
 * What slicer is being used?
 * A full log would be nice or at least answers to the questions above
 * I guess it happens on all prints, otherwise a file that produces the reported behavior 

Again, I'm sorry it took me this long to look into this.

Tigerwraith, 2024-10-05:
I have since moved on from using Marlin2 to Klipper on my Ender. I was using Orca Slicer (I have now moved to Bambu Slicer due to stringing issues using Orca) 

It seemed to me that the ok message was for every single command not just specific ones. 
This is what the creator of the firmware had said to me, 
[image]
I never did get around to compiling the firmware with that advance feature turned off so I do not know if it worked.

DivingDuck, 2024-10-06:
[Marlin doc: ADVANCED_OK](https://marlinfw.org/docs/configuration/configuration.html#advanced-ok):

```
Advanced OK

//#define ADVANCED_OK

Include extra information about the buffer in “ok” messages. Some hosts will have this feature soon.
This could make the NO_TIMEOUTS unnecessary.
```
And here is a information for a OctoPrint plugin that uses this output.
Mentioning: https://community.octoprint.org/t/a-list-of-recommended-marlin-features/39048
Plugin: https://github.com/chendo/BufferBuddy

rockstorm101, 2024-10-06:
> It seemed to me that the ok message was for every single command not just specific ones.
> This is what the creator of the firmware had said to me,

Aha! Finally some light regarding the mysterious `ok N#### P# B#` messages. It's a Marlin thing. I'd say we definitely need to handle those better, otherwise we'll get more issues on this regard.

> I have since moved on from using Marlin2 to Klipper on my Ender.

I assume you don't see the messages any more then?

Tigerwraith, 2024-10-06:
Well no I don't as I don't use pronterface any more. Klipper and Mainsail now.

DivingDuck, 2025-02-12:
@rockstorm101,
Guess, we can close this issue?

rockstorm101, 2025-02-13:
@DivingDuck, I would leave this open until we correctly handle the `ok N#### P# B#` messages. IMHO, those should not be displayed to the user as the OP reported. I'm surprised we didn't get more reports like this tbh. I currently have no printer at hand to test. Do you own a MarlinFW-powered device to reproduce it? Do we already handle those correctly? Could be closed as a "won't fix" if we don't think this is something for Printrun to be concerned with.

DivingDuck, 2025-02-14:
@rockstorm101, 
Unfortunately I have only a one printer with an ancient Marlin 1.0 and this do not show these messages if I remember correct. The other printer run on  Smoothieware which do not show this behavior as well. So I'm not able to reproduce it.
In my second post I highlighted that there is a setting in Marlin for debug levels and one need to check this. We never got an answer from the thread owner whether this is the case or not. We only know that he uses an other firmware/printer now.

I too miss reports from other users in this regard, thought that a lot of printers have Marlin firmware. Agreed, we should close the issue with a "won't fix". One can open a new issue report if there is a problem left. We then get maybe better information.