kliment/Printrun #1444: During the execution of the gcode sequence, one gcode is not executed

equalsai, 2024-08-22:
I excute gcode like that, The firmware uses marlin2.0:

G0 Z10 S0
G90
G21
G1 F1000
G1  X122.4164 Y129.3071
G4 P0 
G0 Z0 S0
G4 P0
G1 F1000.000000
G1  X173.5241 Y129.3071
G1  X173.5241 Y64.5357
G1  X169.0267 Y64.5357
G1  X169.0267 Y68.0069
G1  X126.9139 Y68.0069
G1  X126.9139 Y64.4663
G1  X122.4164 Y64.4663
G1  X122.4164 Y129.3071
G4 P0 
G0 Z10 S0
G1 F1000
G1  X108.5735 Y112.7845
G4 P0 
G0 Z0 S0
G4 P0
G1 F1000.000000
G1  X90.642 Y112.7845
G1  X90.642 Y95.1511
G1  X108.5735 Y95.1511
G1  X108.5735 Y112.7845

I found that the gcode command <G1  X90.642 Y95.1511> was not executed.
I monitored the serial port and got the following results:
This instruction does not return the content of mm=
> COM5,Wirte(28): N26 G1  X90.642 Y95.1511*12\LF
> COM5, Read(1): o
> COM5, Read(1): k
> COM5, Read(1): \LF
> COM5, Wirte(29): N27 G1  X108.5735 Y95.1511*9\LF
> COM5, Read(1): m
> COM5, Read(1): m
> COM5, Read(1): = 
> COM5, Read(1): 2
> COM5, Read(1): 5
> COM5, Read(1): .

DivingDuck, 2024-08-22:
Hi @equalsai,

Thanks for your report. Please add as well the version of Pronterface and your operating system to your report.

Maybe I oversee something, but can you tell me what command is not executed?

I copied your code and executed it on my system (firmware smoothieware) as there is nothing special on your gcode.

<details><summary> Show test gcode </summary>

```
G0 Z10 S0
G90
G21
G1 F1000
G1 X122.4164 Y129.3071
G4 P0
G0 Z0 S0
G4 P0
G1 F1000.000000
G1 X173.5241 Y129.3071
G1 X173.5241 Y64.5357
G1 X169.0267 Y64.5357
G1 X169.0267 Y68.0069
G1 X126.9139 Y68.0069
G1 X126.9139 Y64.4663
G1 X122.4164 Y64.4663
G1 X122.4164 Y129.3071
G4 P0
G0 Z10 S0
G1 F1000
G1 X108.5735 Y112.7845
G4 P0
G0 Z0 S0
G4 P0
G1 F1000.000000
G1 X90.642 Y112.7845
G1 X90.642 Y95.1511
G1 X108.5735 Y95.1511
G1 X108.5735 Y112.7845
```
</details>

This is the communication log:

<details><summary> Show log </summary>

```
2024-08-22 09:21:35,977 - Druck gestartet bei: 09:21:35
2024-08-22 09:21:35,979 - SENT: N0 G0 Z10 S0*17
2024-08-22 09:21:35,982 - RECV: ok T:22.6 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:20.9 /0.0 @0
2024-08-22 09:21:35,983 - RECV: ok
2024-08-22 09:21:35,992 - SENT: N1 G90*17
2024-08-22 09:21:35,993 - RECV: ok
2024-08-22 09:21:36,008 - SENT: N2 G21*24
2024-08-22 09:21:36,009 - RECV: ok
2024-08-22 09:21:36,024 - SENT: N3 G1 F1000*76
2024-08-22 09:21:36,025 - RECV: ok
2024-08-22 09:21:36,026 - SENT: N4 G1 X122.4164 Y129.3071*36
2024-08-22 09:21:36,040 - RECV: ok
2024-08-22 09:21:36,056 - SENT: N5 G4 P0*104
2024-08-22 09:21:44,914 - RECV: ok
2024-08-22 09:21:44,915 - SENT: M117 20.69% Est 0.00.24
2024-08-22 09:21:44,916 - RECV: ok
2024-08-22 09:21:44,930 - SENT: M105
2024-08-22 09:21:44,936 - RECV: ok T:22.6 /0.0 @0 T1:22.1 /0.0 @0 B:21.5 /0.0 @0 B1:21.0 /0.0 @0
2024-08-22 09:21:44,946 - SENT: N6 G0 Z0 S0*38
2024-08-22 09:21:45,512 - RECV: ok
2024-08-22 09:21:45,521 - SENT: M105
2024-08-22 09:21:45,527 - RECV: ok T:22.7 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:21.0 /0.0 @0
2024-08-22 09:21:45,536 - SENT: N7 G4 P0*106
2024-08-22 09:21:45,537 - RECV: ok
2024-08-22 09:21:45,552 - SENT: N8 G1 F1000.000000*105
2024-08-22 09:21:45,554 - RECV: ok
2024-08-22 09:21:45,568 - SENT: N9 G1 X173.5241 Y129.3071*40
2024-08-22 09:21:45,571 - RECV: ok
2024-08-22 09:21:45,584 - SENT: N10 G1 X173.5241 Y64.5357*41
2024-08-22 09:21:50,028 - RECV: ok
2024-08-22 09:21:50,039 - SENT: M105
2024-08-22 09:21:53,910 - RECV: ok T:22.7 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:21.0 /0.0 @0
2024-08-22 09:21:53,923 - SENT: M117 37.93% Est 0.00.19
2024-08-22 09:21:53,924 - RECV: ok
2024-08-22 09:21:53,939 - SENT: N11 G1 X169.0267 Y64.5357*34
2024-08-22 09:21:53,941 - RECV: ok
2024-08-22 09:21:53,955 - SENT: N12 G1 X169.0267 Y68.0069*38
2024-08-22 09:21:54,206 - RECV: ok
2024-08-22 09:21:54,218 - SENT: N13 G1 X126.9139 Y68.0069*45
2024-08-22 09:21:54,446 - RECV: ok
2024-08-22 09:21:54,452 - SENT: N14 G1 X126.9139 Y64.4663*46
2024-08-22 09:21:56,959 - RECV: ok
2024-08-22 09:21:56,961 - SENT: M105
2024-08-22 09:21:57,198 - RECV: ok T:22.7 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:21.0 /0.0 @0
2024-08-22 09:21:57,211 - SENT: N15 G1 X122.4164 Y64.4663*46
2024-08-22 09:21:57,214 - RECV: ok
2024-08-22 09:21:57,226 - SENT: N16 G1 X122.4164 Y129.3071*23
2024-08-22 09:21:57,490 - RECV: ok
2024-08-22 09:21:57,491 - SENT: N17 G4 P0*91
2024-08-22 09:22:01,320 - RECV: ok
2024-08-22 09:22:01,330 - SENT: M105
2024-08-22 09:22:01,335 - RECV: ok T:22.7 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:21.1 /0.0 @0
2024-08-22 09:22:01,345 - SENT: N18 G0 Z10 S0*40
2024-08-22 09:22:01,920 - RECV: ok
2024-08-22 09:22:01,922 - SENT: N19 G1 F1000*119
2024-08-22 09:22:01,924 - RECV: ok
2024-08-22 09:22:01,937 - SENT: N20 G1 X108.5735 Y112.7845*26
2024-08-22 09:22:01,940 - RECV: ok
2024-08-22 09:22:01,952 - SENT: N21 G4 P0*94
2024-08-22 09:22:03,385 - RECV: ok
2024-08-22 09:22:03,386 - SENT: N22 G0 Z0 S0*16
2024-08-22 09:22:05,253 - RECV: ok
2024-08-22 09:22:05,256 - SENT: M117 79.31% Est 0.00.05
2024-08-22 09:22:05,257 - RECV: ok
2024-08-22 09:22:05,272 - SENT: M105
2024-08-22 09:22:05,278 - RECV: ok T:22.7 /0.0 @0 T1:22.0 /0.0 @0 B:21.6 /0.0 @0 B1:21.1 /0.0 @0
2024-08-22 09:22:05,287 - SENT: N23 G4 P0*92
2024-08-22 09:22:05,289 - RECV: ok
2024-08-22 09:22:05,302 - SENT: N24 G1 F1000.000000*87
2024-08-22 09:22:05,303 - RECV: ok
2024-08-22 09:22:05,303 - SENT: N25 G1 X90.642 Y112.7845*27
2024-08-22 09:22:05,307 - RECV: ok
2024-08-22 09:22:05,318 - SENT: N26 G1 X90.642 Y95.1511*44
2024-08-22 09:22:06,317 - RECV: ok
2024-08-22 09:22:06,332 - SENT: N27 G1 X108.5735 Y95.1511*41
2024-08-22 09:22:08,805 - RECV: ok
2024-08-22 09:22:08,808 - SENT: M105
2024-08-22 09:22:09,885 - RECV: ok T:22.7 /0.0 @0 T1:22.1 /0.0 @0 B:21.5 /0.0 @0 B1:21.0 /0.0 @0
2024-08-22 09:22:09,901 - SENT: N28 G1 X108.5735 Y112.7845*18
2024-08-22 09:22:09,904 - RECV: ok
2024-08-22 09:22:09,916 - SENT: M110 N-1
2024-08-22 09:22:09,917 - Druck beendet um: 09:22:09 und dauerte 0:00:33
2024-08-22 09:22:10,207 - SENT: M117 Finished Print
2024-08-22 09:22:10,369 - SENT: M105
2024-08-22 09:22:10,950 - RECV: ok
2024-08-22 09:22:10,951 - RECV: ok

```
</details>

I can't see any error or unusual behavior in Pronterface.

[image]

Looking at your log I wonder from what the output below N27 is. In N27 you make a simple G1 movement. There should be no communication from your machine back to Pronterface except a received of <ok>.

Btw, you can log the communication directly within Pronterface. -->Settings -->Debug Communications. There is also a log file available at your home directory. The file is called printrun.log. Under -->Settings -->Options -->User Interface you can find the location as well.

Test file, log and screenshot: [TestFilePic&Log.zip](https://github.com/user-attachments/files/16707805/TestFilePic.Log.zip)

DivingDuck, 2024-09-12:
Any update @equalsai ?

DivingDuck, 2024-09-14:
I will close the issue as we got no feedback and the issue isn't reproducible.

closed 2024-09-14 by DivingDuck