kliment/Printrun #1446: Malware detection by VirusTotal

DaveInCaz, 2024-09-22:
I scanned file `printrun-2.1.0_windows_x64_py3.10.zip` from  https://github.com/kliment/Printrun/releases/tag/printrun-2.1.0 in [VirusTotal](https://www.virustotal.com/gui/file/34bfe4a0b5784ee4365cc28c7af73d2d062f145a964cbff36575f965e93d5b3a/detection) and it reported some malware issues:

[image]

(It only shows 2 line items here although above that it says "3/70 security vendors flagged this file as malicious")

I then [checked file `Pronterface.exe`](https://www.virustotal.com/gui/file/b8bcf2e8097632db89bc4330e3221c34f26ae3ed32a93c6624744dff553d04f7) from within the ZIP and it showed a bunch more:
[image]

I know false positives can occur but I have no way to tell one way or the other. Thanks

DivingDuck, 2024-09-23:
@DaveInCaz 

I just download the zip file in question. I can't find any virus or malware in our download files for windows. It is still the same file as it was build in Mai 28.

Reports from online VirusTotal scan today:
Windows x64 version:
https://www.virustotal.com/gui/url/e8276f37b6f86b158cec54f3273dff19847315471d5d9a9aaacea669baee377f
Windows x86 version:
https://www.virustotal.com/gui/url/af83cd65c418d02febbba3fa7f35b743b880f712ce1da423e20a7fb66fab83a8

We have actually  28k downloads of this file and no claims for a virus / malware infection, so my best answer is that this is a false positive in case you download the installation files from our repository https://github.com/kliment/Printrun/releases (this is the only place we publish official versions)

Additional info: 
https://github.com/kliment/Printrun/issues/1312#issuecomment-2351691683

rockstorm101, 2025-03-21:
Closing since it wasn't a bug and it was addressed.

closed 2025-03-21 by rockstorm101