kliment/Printrun #1447: crearing loop buton in pronterface

lesmainsdanslamachine, 2024-09-25:
Hello,

I'm starting to use pronterface to control printing. I'm creating buttons for certain operations that I need to correct during printing. I would like to create a button that allows the same code to be sent continuously,
For example: G1 Z0.5 F60
This way, if ja'cticve.le.voiton, the code is sent in a loop. Ideally, I'd also like to create a slider to vary the a.
Is this possible with pronterface?
Can anyone help me?
Merci par avance

rockstorm101, 2024-10-05:
Hi @lesmainsdanslamachine, as far as I know, this would not be possible within Pronterface via buttons. They were meant to perform their task only once.

If you need to continuously correct stuff during the print it sounds to me like what you need to correct is the G-code you are using. I mean, maybe inject these operations you need into the G-code file itself so that they are performed as part of the print instead of making Pronterface intervene the print all the time. For instance, on PrusaSlicer, you can custom G-code at layer changes [[1]].

[1]: https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490

lesmainsdanslamachine, 2024-10-05:
Thank you for your reply.
Yes, I do try to send code directly to the printer, no error sylptome from a previous code, there is none.
My current solution uses pronterface and autohotkey which allows me to send code, generate keys dedicated to specific gcode, and make loops.
I've been using it for a week.
The next step will be to dispense with a computer and connect a keyboard directly to the printer, perhaps via arduino.