kliment/Printrun #1448: Connection error

jessepietila, 2024-09-25:
[image]

Ender  3 Pro with Pronterface,many laeyr fine then  stops  printing
[image]

What to do to fix ?

DivingDuck, 2024-09-25:
Hi,
It looks like you send an unknown command "M73 Pxx" and your printer tells you that he didn't understand this command (see first 3 lines). 

More concerns me that your com port is locked (see "Access denied"). Did you run a slicer or an other program parallel that try to communicate thru the same com port? Cura is known to lock com ports when you start the application.

Btw. , what Pronterface version and what operating system are we talking about?

jessepietila, 2024-09-25:
Kind thanks DivingDuck,

it was moreover the  Pronterface.. which did send the M73 Pxx command, I did not do it...

I did not run Cura... or maybe it was open...  usually I start Prionterface first then Cura to access printer. Another way around can not access Cura... as I assume You to know

I did run Pronterface 2.0 and 2.1.0

Looking forward to here more of Your advices thank You !

DivingDuck, 2024-09-25:
> it was moreover the Pronterface.. which did send the M73 Pxx command, I did not do it...

Pronterface send what ever gcode is in the gcode file you try to print. Pronterface itself do not add any additional code to an existing print. The only way you can add gcode commands during a print with Pronterface is by entering commands in the console window or injecting via macro.

>... usually I start Prionterface first then Cura to access printer ...
Well, it happen to me in the past (long, long time ago). The better way is to deactivate the com plugin in Cura if you know you want to use Cura during printing.

[image]

Use Pronterface 2.1, the older versions have some problems that are solved in the latest version.

Can you attach the log file (make a zip file and attach it here)? Maybe it show some more information. On Windows it is called `printrun.log` and you will find it in your root user path like `c:\Users\YOUR_USER_NAME\printrun.log`

jessepietila, 2024-09-25:
Ok!

[printrun.zip](https://github.com/user-attachments/files/17138908/printrun.zip)

zipped printrun.log above

ok, so I will enable the USB Printing 1..0.2. via marketplace?

DivingDuck, 2024-09-25:
> ok, so I will enable the USB Printing 1..0.2. via marketplace?

No, you need to disable it. Enable is shown if the plugin is disabled. 🙂 And yes via market place. Use the gearwheel (top right on the window as marked in my picture)

Regarding logfile, this is your last print attempt:

```
2024-09-25 20:24:48,927 - Loading file: C:\Users\jesse\Documents\Lippa_23\Lippa_yksviiteen\CFFFP_Lippa111_yksviiteen_t.gcode
2024-09-25 20:24:49,442 - Loaded C:\Users\jesse\Documents\Lippa_23\Lippa_yksviiteen\CFFFP_Lippa111_yksviiteen_t.gcode, 35429 lines
2024-09-25 20:24:49,446 - 6.55m of filament used in this print
2024-09-25 20:24:49,446 - The print goes:
2024-09-25 20:24:49,446 - - from 0.00 mm to 165.70 mm in X and is 165.70 mm wide
2024-09-25 20:24:49,446 - - from 0.00 mm to 130.69 mm in Y and is 130.69 mm deep
2024-09-25 20:24:49,446 - - from 0.00 mm to 59.00 mm in Z and is 59.00 mm high
2024-09-25 20:24:49,446 - Estimated duration: 196 layers, 1:29:13
2024-09-25 20:24:57,000 - Setting hotend temperature to 215 degrees Celsius.
2024-09-25 20:24:57,888 - Setting bed temperature to 55 degrees Celsius.
2024-09-25 20:24:59,465 - Print started at: 20:24:59
2024-09-25 20:24:59,471 - echo:Unknown command: "M73 P0"
2024-09-25 20:30:08,051 - echo:Unknown command: "M73 P1"
2024-09-25 20:30:40,520 - echo:Unknown command: "M73 P2"
2024-09-25 20:31:42,931 - echo:Unknown command: "M73 P3"
2024-09-25 20:32:34,870 - echo:Unknown command: "M73 P4"
2024-09-25 20:33:08,420 - echo:Unknown command: "M73 P5"
2024-09-25 20:34:14,833 - echo:Unknown command: "M73 P6"
2024-09-25 20:35:13,840 - Welcome to the printer console! Type "help" for a list of available commands.
2024-09-25 20:35:23,851 - echo:Unknown command: "M73 P7"
2024-09-25 20:36:15,101 - echo:Unknown command: "M73 P8"
2024-09-25 20:36:51,943 - echo:Unknown command: "M73 P9"
2024-09-25 20:37:54,557 - echo:Unknown command: "M73 P10"
2024-09-25 20:38:39,071 - echo:Unknown command: "M73 P11"
2024-09-25 20:39:47,351 - echo:Unknown command: "M73 P12"
2024-09-25 20:40:33,857 - echo:Unknown command: "M73 P13"
2024-09-25 20:42:07,284 - Print paused at: 20:42:07
2024-09-25 20:42:07,287 - Can't write to printer (disconnected?)Unable to write to serial port '{self.port}'
2024-09-25 20:42:16,445 - Resuming.
2024-09-25 20:42:42,088 - Setting hotend temperature to 0 degrees Celsius.
2024-09-25 20:42:42,648 - Setting bed temperature to 0 degrees Celsius.
2024-09-25 20:42:43,768 - Setting hotend temperature to 0 degrees Celsius.
2024-09-25 20:42:54,497 - Disconnecting from printer...
2024-09-25 20:42:54,498 - Exiting program. Goodbye!
2024-09-25 23:38:33,993 - Not connected to printer.
2024-09-25 23:38:33,994 - Disconnecting from printer...
2024-09-25 23:38:33,994 - Exiting program. Goodbye!
```
And a little but above:

```
2024-09-25 20:24:09,406 - Got rubbish reply from COM4 at baudrate 250000:
Maybe a bad baudrate?
```

Some remarks:
Your baud rate seems to be to fast, try a slower rate `115200`. I saw often that users had problems with the higher baud rates, esspecially if the cabe is long and not from good quality. Or there are interference signals that disrupt the communication signals.

As I already mentioned, your gcode file includes `M73` commands that your printer do not understand. You need to delete those gcode lines.

`Print paused at: 20:42:07` This means you have paused the print. Pronterface then loose the connection with your printer. Maybe the problem is the too fast com baud rate too. You need to check this.

jessepietila, 2024-09-26:
ok thaanks,

understood.  Still do not understand why my gcode includes M73 codes...? ..and how to remove them..?

USB printing is and is been unabled. long cable yes I will be cautious  with baud rate.

I have used Pronterface succesfully many  years, this is first time with problems / Ender3Pro

DivingDuck, 2024-09-26:
> Still do not understand why my gcode includes M73 codes...? ..and how to remove them..?

This is usually a setting in the slicer. In Cura you can do this via post processing and the scrip Display Info on LCD where you can add M73 (Cura menu: Extensions -->Post Processing -->Modify Gcode -->Add Script -->Display Info on LCD)

An other option is maybe updating the firmware and enable M73

jessepietila, 2024-09-26:
ok kind thanks !!

DivingDuck, 2024-10-03:
Hi @jessepietila,
This seems to be not a bug in Pronterface. Was you able to solve your problem and can we close the issue?

jessepietila, 2024-10-03:
Yes solved, thank You!

DivingDuck, 2024-10-03:
Thanks for your response and happy printing. I will close the issue.
Best regards, DD

closed 2024-10-03 by DivingDuck