kliment/Printrun #1449: Blocking versions of `send` and `send_now` do not exist

tomasiser, 2024-09-27:
Hello, in Printrun 2.1.0, there are `send` and `send_now` methods as [part of `printcore.py`](https://github.com/kliment/Printrun/blob/master/printrun/printcore.py#L520).

These methods are **non-blocking**, which means when you send a `G0 ...` command to move to a certain position, the method returns immediately without waiting for the printer to finish the movement.

Could blocking versions of these commands be implemented? Is it even possible given the communication protocol? Of course you can calculate a rough estimate of the time it needs to move by dividing the distance by the feedrate, but that is not the same as a blocking version of these methods.

kliment, 2024-09-27:
There are no blocking methods like this. However, there is a synchronization command M400. M400 will not complete until all moves are done, so sending two M400s in a row will let you check for completion by seeing if queue is empty.

For related work, see https://github.com/kliment/Printrun/issues/990 and https://github.com/kliment/Printrun/pull/1008 which implements a very similar thing but needs to be rebased because it's drifted away from the current state of the code.

tomasiser, 2024-10-02:
Thank you, so what you recommend as a workaround is to send `M400` twice and then wait until `priqueue.empty()`? :)

kliment, 2024-10-03:
That's probably the easiest way to do it, yes.