kliment/Printrun #1453: imghdr will be deprecated soon

rockstorm101, 2024-10-12:
Originally [reported by Louis-Philippe Véronneau][1].

`imghdr` [will be deprecated][2] with Python 3.13. A replacement could be [`filetype`][3], though the API is completely different so it will require some work.

[1]: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1084669#5
[2]: https://peps.python.org/pep-0594/#imghdr
[3]: https://pypi.org/project/filetype

DivingDuck, 2024-10-12:
Hi,

Looking at https://peps.python.org/pep-0594/#imghdr and the recommended replacement alternatives I tend a bit more to [puremagic](https://pypi.org/project/puremagic/). It looks like a easy usable drop-in replacement. In addition it seems to be a living project. What ever this means...

With [filetype](https://pypi.org/project/filetype/) and [python-magic](https://pypi.org/project/python-magic/) I have mixed feelings as both don't look like active projects. Python support is shown up to version 3.9.

What do you think about it?

DivingDuck, 2024-10-12:
This is a quick one: PR#1455

Test needed for Linux and macOS version of pronterface.

Artifacts:
Windows: https://github.com/kliment/Printrun/actions/runs/11306128512
macOS: https://github.com/kliment/Printrun/actions/runs/11306128506
Python sdist and wheels: https://github.com/kliment/Printrun/actions/runs/11306128509

Test files used in Pronterface -->Tools -->Projector: [ProjectorTestFiles.zip](https://github.com/user-attachments/files/17350246/ProjectorTestFiles.zip)
One is for .svg format and one for Prusa .sl1s format

neofelis2X, 2024-10-12:
Hello,

that was fast! I would have suggested that we can remove imghdr and just look at the file extensions. Because in this usecase the images should always have a correct file extension (.png in most cases). But this solution is good too!

I tested it on macOS and it works fine.

rockstorm101, 2024-10-12:
> Looking at https://peps.python.org/pep-0594/#imghdr and the recommended replacement alternatives I tend a bit more to [puremagic](https://pypi.org/project/puremagic/). It looks like a easy usable [...]

I did not investigate other alternatives really. `puremagic` looks like it has a closer API to what we had so I think you are right and it is the best bet.

> Test files used in Pronterface -->Tools -->Projector: [ProjectorTestFiles.zip](https://github.com/user-attachments/files/17350246/ProjectorTestFiles.zip)
> One is for .svg format and one for Prusa .sl1s format

I did a quick test on Linux though I don't have a resin printer or the like. No errors thrown so it looks OK to me. I'm happy to merge #1455 if you want.

DivingDuck, 2024-10-12:
Then we can merge it. Thanks for testing 🙂

commit 559109a7 (2024-10-13): feat: Replace module imghdr with puremagic (#1455)

`imghdr` becomes obsolete with Python 3.13

Fixes #1453
closed 2024-10-13 by rockstorm101