kliment/Printrun #1456: Pip will not install dependencies

rockstorm101, 2024-10-12:
https://github.com/kliment/Printrun/blob/d4080ef32cfdee0cc76a4d633dd0a1f8b1491425/README.md?plain=1#L169-L175

According to the readme instructions the command `python -m pip install .` would install Printrun and its dependencies. However the dependencies are not being installed.

See fresh run using Python 3.12.7 and pip 24.2. Only Printrun gets installed now:
```shell
$ python3 -m venv venv
$ source venv/bin/activate
$ python -m pip install --dry-run .
Processing /home/rock/Projects/printrun/rockintrun
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Would install Printrun-2.1.0
```

I would have sworn that the above command used to install everything. I've got the fix already prepared but I wondered if someone else stumbled across this before.

rockstorm101, 2024-10-12:
Same thing if I install from PyPI:

```shell
$ python3 -m venv venv
$ source venv/bin/activate
$ python -m pip install --dry-run --no-cache-dir Printrun
Collecting Printrun
  Downloading Printrun-2.1.0-cp312-cp312-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (25 kB)
Downloading Printrun-2.1.0-cp312-cp312-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (602 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 602.3/602.3 kB 13.8 MB/s eta 0:00:00
Would install Printrun-2.1.0
```

DivingDuck, 2024-10-12:
My last "experiment" with Debian on a VM ends in a disaster. I thought I'm not compatible with Debian and deleted the VM for not getting more frustrated.

PR #1457 (2024-10-13): build: Fix run-time dependencies installation
Fixes #1456
commit 6e005a3c (2024-10-13): build: Fix run-time dependencies installation (#1457)

Fixes #1456
closed 2024-10-13 by rockstorm101