kliment/Printrun #1458: Pronterface caused a bluescreen and now unable to open after reinstall

Sleff1, 2024-10-15:
When i opened pronterface and connected my printer via usb cable I got a bluescreen and now i can't open the program. Reinstalling didn't help. 

Error i get:
```
Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2596, in __init__
  File "printrun\pronterface.py", line 171, in __init__
  File "printrun\pronsole.py", line 778, in parse_cmdline
  File "printrun\pronterface.py", line 1161, in process_cmdline_arguments
  File "printrun\pronsole.py", line 761, in process_cmdline_arguments
  File "printrun\pronsole.py", line 656, in load_default_rc
  File "printrun\pronsole.py", line 627, in load_rc
  File "cmd.py", line 211, in onecmd
  File "printrun\pronsole.py", line 403, in default
  File "cmd.py", line 236, in default
AttributeError: 'NoneType' object has no attribute 'write'
```

DivingDuck, 2024-10-15:
What version of Pronterface is installed?

Sleff1, 2024-10-15:
2.1.0 x64

DivingDuck, 2024-10-15:
This is unusual. Can you try to reboot your system?

Sleff1, 2024-10-15:
I rebooted the system and the same error shows up and I also don't have printrunconf.ini and I only have this log:
[printrun.log](https://github.com/user-attachments/files/17381276/printrun.log)

DivingDuck, 2024-10-15:
Looking to the log I see you have set a very high baud rate for connecting to your printer. Depending on the environment, length and quality of your usb cable you you may need to check what rate is needed for a save communication. Try to use a lower one, start with: 115200 baud.

Sleff1, 2024-10-15:
I set the baud rate that high because I had a klipper flashed on the printer that was set that high. I was just checking if I would get any response because my laptop only connects to the mcu when the printer power supply is turned off and I don't know why(pc connects only sometimes looks like it's random).

DivingDuck, 2024-10-15:
This is why I mentioned a much lower baud rate, because I see in your log file that Pronterface isn't able to establish a robust connection with your setup. Please start with 115200 baud.

Sleff1, 2024-10-15:
I would but I cannot open pronterface due to that error even after reinstalling and deleting the  log

DivingDuck, 2024-10-15:
Search for the file printrunconf.ini on your computer. On windows 10 there are two possible locations:
one is the already mentioned location `c:\Users\YourUserName\printrun.log` and the other is `c:\Users\YourUserName\AppData\Local\Printrun\`, where you need to change `YourUserName` with your real Windows Username.

Keep in mind that the folder AppData is a hidden directory.

Sleff1, 2024-10-16:
I deleted that file and now it works. Thank you so much for your help.

DivingDuck, 2024-10-16:
Perfect. Good to hear. Now you need to check baud rate would work best for your usb cable and printer.

BTW, where was the file located?

DivingDuck, 2024-10-17:
@Sleff1,
I will close the issue as your Pronterface installation is now working.
Best regards,
DD

closed 2024-10-17 by DivingDuck