kliment/Printrun #1462: PRONTERFACE reports error message "AttributeError: 'NoneType' object has no attribute 'write' "

FredBello, 2024-10-29:
Hello,
I can't open PRONTERFACE, even using the latest version. 
WINDOWS 10 64 BITS 
PRONTERFACE: printrun-2.2.0_windows_x64_py3.10

The following error message appears:
"Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
  File "printrun\pronterface.py", line 2596, in __init__
  File "printrun\pronterface.py", line 171, in __init__
  File "printrun\pronsole.py", line 778, in parse_cmdline
  File "printrun\pronterface.py", line 1161, in process_cmdline_arguments
  File "printrun\pronsole.py", line 761, in process_cmdline_arguments
  File "printrun\pronsole.py", line 656, in load_default_rc
  File "printrun\pronsole.py", line 627, in load_rc
  File "cmd.py", line 211, in onecmd
  File "printrun\pronsole.py", line 403, in default
  File "cmd.py", line 236, in default
AttributeError: 'NoneType' object has no attribute 'write'   "

I don't know what to do, could you give me some suggestions?
Thanks

DivingDuck, 2024-10-29:
Hi @FredBello,
this looks like you have a malformed `printrunconf.ini`. Close Pronterface and try to rename or delete that file and then start it again.
Please check the file `printrun.log` too. You can add the file here in a zipped version for further investigation if needed.

You should find both files in your root user folder, like `c:\Users\YOUR_USER_NAME\` where `YOUR_USER_NAME` is your windows user name. Please check the other possible location at `c:\Users\\YOUR_USER_NAME\AppData\Local\Printrun\Printrun\` in case you can't find them on the first place.
 
Remark: ...\Appdata\... is a hidden folder, you may want to allow showing hidden files and folders in the file explorer.

FredBello, 2024-10-30:
Hello, Diving Duck
   Your instructions were spot on. After deleting the files you mentioned, the program started again.
   I would like to thank you very much for your attention and generosity in giving your time to help, thank you very much!

God bless you and your family!
Hugs from Brazil

closed 2024-10-30 by FredBello

DivingDuck, 2024-10-30:
You are welcome. Glad to hear that it works for you. And thanks for the feedback.

All the best to you and your family, 
DD