kliment/Printrun #1464: Move Right Custom Button.

igabigi, 2024-11-09:
The cause of the issue is I created a custom button and then right-clicked to move-left and it disappeared, the next time I opened the program I had an error. I tried reinstalling from binaries, from source. Nothing works

When I open Pronsole after re-installing, I still get : 

WARNING:root:Memory-efficient GCoder implementation unavailable: cannot import name 'gcoder_line' from 'printrun' (C:\Users\username\AppData\Local\Temp\_MEI11202\printrun\__init__.pyc)
*** Unknown syntax: button 1 "GET CURRENT POSITION" /c "green" M114
Welcome to the printer console! Type "help" for a list of available commands.
offline>

What I am understanding is that the button is still in my pc memory but I am not sure where to access it to delete it. 

[image]

igabigi, 2024-11-09:
Okay I found the path C:\Users\user\AppData\Local\Printrun\Printrun\printrunconf.ini

rockstorm101, 2024-11-10:
Duplicate of #1404

closed 2024-11-10 by rockstorm101