kliment/Printrun #1466: The M503 command does not work

MrTashy, 2024-11-17:
Greetings, please tell me, there is a problem on the geeetech a30m 3d printer when entering a command in the Pronterface program, it writes sending and that's it... How to solve this problem?
[image]

DivingDuck, 2024-11-18:
Hi,
it looks like you have a bad connection between your printer and computer. Maybe your USB cable is too long for the baud rate you choose. Try it with a shorter cable and/or a slower baud rate.
What Pronterface version and operating system are you running?
What is this "JumpToApp0" and "JumpToApp1"? This is defiantly nothing what came from Pronterface.

MrTashy, 2024-11-18:
The thing is, I'm running a pdi calibration. And he does it. Plus, they are displayed correctly at all temperatures and I can move the table and the extruder through the panel. Version of Pronterface 2.2.0. Windows 10

DivingDuck, 2024-11-18:
Well, if Pronterface tells it got rubbish replays then there is nothing it can do against it. M503 is a valid gcode and after sending Pronterface should receive the override settings from your printer. 

Not receiving the data means that either Pronterface lost the connection to the printer (what initialize a error report like `Can't read from printer (disconnected?) Unable to read from serial port 'com7' ` and/or `Can't write to printer (disconnected?)Unable to write to serial port '{self.port}' ` or the printer maybe hangs (check if you can send other commands after you send M503 and don't got a response)

My concern is that your printer answers it don't know the command M503 as shown in your picture. I haven't seen a printer so fare that did not understand this command. What happen if you send commands like M114, M115, M119?

I don't think that Pronterface is the cause of your problem when you can run PID tuning w/o a problem.

MrTashy, 2024-11-18:
[image]
[printrun.zip](https://github.com/user-attachments/files/17799299/printrun.zip)

MrTashy, 2024-11-18:
Кабель длиной один метр. В другие usb порта пробовал подключать-безуспешно. 
[image]

MrTashy, 2024-11-18:
The cable is one meter long. I have tried connecting to other ports, but to no avail. I attach photos of attempts to connect to other speeds.
[image]

DivingDuck, 2024-11-18:
1 m should be ok if the cable isn't from really bad quality. Have you a other USB cable on hand for testing?

Pls. attach the log file once more. It show more information as the screenshot.

One more thing:

Is it maybe possible that your printer stops because of too long inactivity? What happen if you run the M999 command?

With -->Settings -->debug communications you can try to get a bit more info's about what happens with the printer communication

An other way to test the serial connection is to use a terminal app like Putty. Set the com port (in  your case COM4) and baud rate and connect.
In the terminal window you can send gcode commands as well.

[image]

One more Question: Do you run a slicer like Cura at the same time you run Pronterface?

Now I run out of ideas...

MrTashy, 2024-11-18:
I tried another cable, it didn't help. The M999 command doesn't give out anything either. I downloaded Putty, I'm attaching a screenshot.I couldn't write anything in Putty.
[printrun.zip](https://github.com/user-attachments/files/17803764/printrun.zip)
[image]
[image]

DivingDuck, 2024-11-18:
You have a connection and got some information back, so in the end your serial connection seems to work fine.

Looking in your log file I recognize negative temps for your second extruder:
`2024-11-18 19:43:13,917 - RECV: ok T:145.4 /0.0 B:18.8 /0.0 T0:145.4 /0.0 T1:-19.0 /0.0 T2:-19.0 /0.0 F:0 R:100 @:0 B@:0`

M503 seems to be not available in your firmware. I did just a quick google search and recognize some results for your printer (geeetech a30 w/o m at the end) in a German forum where one user said that M503 is not present in the stock firmware. Maybe it is worth to check if there is a newer firmware for your printer available.

Regarding M999, this code restart the printer after an error had happen. It do not give any response back but after sending the command you can send other commands again
https://reprap.org/wiki/G-code#M999:_Restart_after_being_stopped_by_error

MrTashy, 2024-11-18:
I changed two extruders to one another, so it shows negative temperatures. The extruder is not connected.

MrTashy, 2024-11-19:
The firmware is original. Maybe it makes sense to put another one?

DivingDuck, 2024-11-19:
I don't know. If the printer do what it should do it is only a matter of taste. Do you really need M503?
On the other side, it looks like geeetech provide both versions, a original stock version like you have installed and a newer one
https://www.geeetech.com/download.html?download_id=44, so you can always step back again.

DivingDuck, 2024-11-23:
@MrTashy,
Did you made a decision regarding a firmware update?

MrTashy, 2024-11-23:
I couldn't figure out how to implement it. And I'm not sure that it will help.

DivingDuck, 2024-11-23:
I saw there is a geeetech channel at youtube and a video for your printer type:
https://www.youtube.com/watch?v=tKdZpohbVEA

I think form the view of Pronterface there is nothing it can help with the M503 issue for this firmware except to find a new firmware which support M503.
We should close the issue if there is nothing else where I can give you some support for Pronterface.

MrTashy, 2024-11-23:
I will try to reinstall the firmware

DivingDuck, 2024-11-23:
OK, I leave the issue open then and wait for your feedback of progress 🤞

MrTashy, 2024-11-23:
Hello, I have some very bad news. I tried reinstalling the firmware, but nothing worked. In addition, my LCD display is no longer working

DivingDuck, 2024-11-23:
Oh, indeed bad news.
How did you do the update?

MrTashy, 2024-11-23:
It seems to have found a solution to the problem. The fact was that it was necessary to download the firmware installer program from the official website. An adapter for the SD card will arrive tomorrow to install the firmware.

MrTashy, 2024-11-23:
So far I have installed the firmware on the motherboard, but still the M503 does not work

MrTashy, 2024-11-24:
I updated the firmware, but the M503 command still does not work

MrTashy, 2024-11-24:
Maybe Pronterface doesn't work for Smatto firmware

DivingDuck, 2024-11-24:
Hi,
sorry to hear. Is your display working again?

Regarding Pronterface, I can surely disagree. M503 is a standard gcode command. The only action Pronterface take is sending that gcode to the printer. The rest is done by the printer firmware. 
How this works in general is, that the firmware interpret a received gcode line and confirm the received gcode line (there is a little bit more then that, but that do not matter). The firmware itself decode the received gcode line and process it further, what is in this case send configuration information. The firmware usually send an error message if it do not know a received gcode command, but some do not. Btw, this is why I ask to send that gcode via Putty as well, because that way you have a simple terminal connection w/o further processing and everything what you send and what will be receive is viewable as raw data.

As I already mentioned, Pronterface have no problem with processing that command and nearly all flavors of 3D printer firmware's I'm aware of do support this command, except some closed code firmware where a printer manufacturer actively disable M503.

The only other way to go is setting up a Marlin firmware your self or find someone who did this already for your printer. The later is the way you can try to look further.

MrTashy, 2024-11-24:
Yes, my display is working. As far as I know, Marlin firmware cannot be installed on Geeetech 3d printers

DivingDuck, 2024-12-08:
I close this issue now as there is nothing we can do from our side.
Best regards, DD

closed 2024-12-08 by DivingDuck