kliment/Printrun #1471: Weird visual glitch?

Simen-D37, 2024-12-02:
[image]
[image]

I'm wanting to print these circle tests but on the pronterface viewer I keep getting these monstrosities.
Does anyone know if this is just a visual glitch or if there is possibly something wrong with the g-code. 
I attached the prusa slicer as a way to show what it should be.

DivingDuck, 2024-12-02:
Please attache the gcode file and, if possible, the model too.

Simen-D37, 2024-12-03:
It must've been because I had a friend slice the the same model and put that gcode into my pronterface and it turned out perfect. Anyway thanks for the help

closed 2024-12-03 by Simen-D37

DivingDuck, 2024-12-03:
Thanks for the feedback and happy printing.