kliment/Printrun #1472: Pronterface can't log to stdout

Fat-Zer, 2024-12-04:
subj;

As tooltip in *Settings->Options->User Interface->Log Path* says:

> Path to the log file. An empty path will log to the console.

But specifying an empty string causes pronterface to log to the default path instead (which is home directory).

This happens because for an empty logpath pronterface saves `set log_path` to `.pronsolerc`, which is interpreted as "print log_path" instead of "reset the variable to an empty string".

DivingDuck, 2024-12-04:
Good point.
This is an intended behavior that we introduce with version 2.0 as we had often the problem that users wasn't able to tell us what they really did and/or they never found the log file. Since then we have always a log file available in the home directory except the user choose an other location.
We should delete the second sentence because of the change we did.

Btw, for Windows it is called printrun.log

Fat-Zer, 2024-12-04:
It would be nice to have an option to log to stdout (or better stderr) as well then, as it's easier to track during debug.

commit 271b92f4 (2024-12-31): feat: Add an option to log to stdout (#1473)

Closes #1472
closed 2024-12-31 by rockstorm101