kliment/Printrun #1475: Repeated threat warnings from Avast when attempting to download win64 zip file

rv7charlie, 2024-12-13:
See the popup:
[image]

DivingDuck, 2024-12-13:
https://github.com/kliment/Printrun/wiki/How-to-%3F#malware-virus-or-trojan-detection-defender-smartscreen-and-firewall-warnings

rv7charlie, 2024-12-13:
Thanks; I figured it was a false positive, but wanted  to ask. If it's triggering multiple anti-virus programs, does anyone have any idea *why*?

DivingDuck, 2024-12-13:
No problem.
Regarding why, it looks like Avast don't update there database regular. So it is up to the users to decide what is false positive and what not. 

Looking on the stats, I see the actual Pronterface for Windows version had ~16k downloads since we published it in October, 20th. Compared to other programs with high installation volumes we are quite small but there was enough time for updating databases in case users report a false positive detection.

Anyway,the best way is to check the download link with a trusted external online detector like I did here via VirusTotal (or similar tools) for the actual x64 version:
https://www.virustotal.com/gui/url/0922abd86c76a02159012c75cfb3c454fd445882788e419f5e2d768329ff61e0

DivingDuck, 2024-12-14:
I close the issue this isn't an issue for Pronterface.
Best regards, DD

closed 2024-12-14 by DivingDuck