kliment/Printrun #1476: Running from source in Ubuntu 24.04 - wxPython import error

mahtDFR, 2024-12-28:
## Overview
This details the troubleshooting process for setting up Pronterface (part of Printrun) on Ubuntu 24.04 (Noble). The primary issue was missing dependencies and wxPython import errors, particularly related to `libtiff.so.5`.

---

## Problem Summary
- **System:** Ubuntu 24.04 (Noble)
- **Python Version:** 3.9
- **Virtual Environment:** Python 3.9 virtual environment (venv39)
- **Issue:** Pronterface failed to run due to missing wxPython and system dependencies.

---

## Error Messages
1. **wxPython Import Error:**
```
wxPython >= 4 is not installed. This program requires wxPython >=4 to run.
```
2. **Missing libtiff.so.5:**
```
ImportError: libtiff.so.5: cannot open shared object file: No such file or directory
```

---

## Troubleshooting Process

### Step 1: Install wxPython
```bash
pip install wxpython
```

**Result:**
```
Requirement already satisfied: wxPython==4.2.1
```
Despite the successful installation, Pronterface continued to report that wxPython was not installed.

---

### Step 2: Check wxPython Import Directly
```bash
python -c "import wx; print(wx.__version__)"
```
**Error:**
```
ImportError: libtiff.so.5: cannot open shared object file: No such file or directory
```

---

### Step 3: Identify Missing Dependencies
```bash
ldconfig -p | grep libtiff
```
**Output:**
```
libtiff.so.6 (libc6,x86-64) => /lib/x86_64-linux-gnu/libtiff.so.6
```

**Problem:** `libtiff.so.5` was required, but `libtiff.so.6` was installed.

---

### Step 4: Create Compatibility Symlink
```bash
sudo ln -s /lib/x86_64-linux-gnu/libtiff.so.6 /lib/x86_64-linux-gnu/libtiff.so.5
sudo ldconfig
```

---

### Step 5: Verify Symlink
```bash
ls -l /lib/x86_64-linux-gnu/libtiff.so.5
```
**Output:**
```
libtiff.so.5 -> /lib/x86_64-linux-gnu/libtiff.so.6
```

---

### Step 6: Test wxPython Import
```bash
python -c "import wx; print(wx.__version__)"
```
**Result:**
```
4.2.1
```

---

### Step 7: Run Pronterface
```bash
python pronterface.py
```
**Success:** Pronterface launched successfully.

---

## Conclusion
The root cause was a missing `libtiff.so.5` dependency. Creating a symlink to the available `libtiff.so.6` resolved the issue. This workaround should function until official `libtiff5` packages are made available for Ubuntu 24.04.

---

## Notes
- Ensure all necessary dependencies are installed:
```bash
sudo apt install libtiff-dev libsdl2-2.0-0 libjpeg-turbo8 libpng16-16
```
- If new errors arise, rerun:
```bash
ldconfig -p | grep <missing-library>
```
- Create symlinks carefully to avoid compatibility issues.

rockstorm101, 2024-12-28:
Hi @mahtDFR thanks for such a detailed report. Step 2 shows clearly that the issue belongs to wxPython and not to Printrun strictly speaking. Nonetheless, could you please confirm whether the issue occurs with latest wxPython 4.2.2? Many thanks in advance

rockstorm101, 2025-01-01:
> could you please confirm whether the issue occurs with latest wxPython 4.2.2?

The answer is yes (as tested at [[1]]). Therefore Ubuntu 24.04 (and probably its upstream/downstream distros too) requires wxPython 4.2.2. I'll flag this issue as "Linux-only" because so far the wxPython 4.2.1 has worked fine everywhere else.

[1]: https://github.com/kliment/Printrun/actions/runs/12569943820/job/35039173486

lucianoloder, 2025-05-14:
On linux mint 22 - wilma, based on Ubuntu 24 - noble, with python 3.12.3 at Step 2 i got:

ImportError: /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0: version `WXU_3.2.6' not found

Tried different wheels on https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-24.04/ but only wxpython-4.2.3-cp312-cp312-linux_x86_64.whl was able to install.

Executing ldconfig showed that these libraries exist.

ldconfig -p | grep libwx_baseu-3.2
	libwx_baseu-3.2.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0
	libwx_baseu-3.2.so.0 (libc6) => /lib/i386-linux-gnu/libwx_baseu-3.2.so.0
	libwx_baseu-3.2.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libwx_baseu-3.2.so

Any hints of what could i do?

Tried to remove venv and reinstall it, but apparently it did not worked.

rockstorm101, 2025-05-14:
> only wxpython-4.2.3-cp312-cp312-linux_x86_64.whl was able to install

Do you mean that the wheel `wxPython-4.2.2-cp312-cp312-linux_x86_64.whl` did not install but the one for wxPython 4.2.3 did?

> libwx_baseu-3.2.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0
> libwx_baseu-3.2.so.0 (libc6) => /lib/i386-linux-gnu/libwx_baseu-3.2.so.0

The difference between the two computers seems to be the second line quoted above. The two libraries quoted there are the same but for different architectures 64bit and 32bit.

> Any hints of what could i do?

Could you try uninstalling the library for 32bits (assuming your computer runs on 64bits)?

P.S.: Still looking like an wxPython issue and not one for Printrun but we'll see.

lucianoloder, 2025-05-15:
Hi @rockstorm101 . Thanks for the reply

I was able to install
 wxPython-4.2.2-cp312-cp312-linux_x86_64.whl , but not
 wxPython-4.2.2-cp311-cp311-linux_x86_64.whl or 
wxPython-4.2.2-cp313-cp313-linux_x86_64.whl

I removed some libraries from this PC and now ldconfig only shows 2 libraries for x86-64, and both are symlinks to another:

ldconfig -p | grep libwx_baseu-3.2
	libwx_baseu-3.2.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0
	libwx_baseu-3.2.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libwx_baseu-3.2.so

ls -la /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0
lrwxrwxrwx 1 root root 24 mar 31  2024 /lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0 -> libwx_baseu-3.2.so.0.2.2

It might be a problem with my installation, since i copied the pronterface folder from the working PC to the faulty one and still have the same issue.

I dont know how to debug it further. It seems that python is missing this library, but i dont know how to fix it. Is it a python3 problem? a ldconfig one? Is there a problem with this virtual environment? How can i remove everything and do a fresh install? Is it sufficient to remove the venv folder?

In the initial installation i ran "python -m pip install ." before " python -m pip install https://extras.wxpython.org/wxPython4/extras/linux/*" , and in the working PC i did backwards. Can this cause this problem?

Thanks in advance.

rockstorm101, 2025-05-15:
> I was able to install
> wxPython-4.2.2-cp312-cp312-linux_x86_64.whl , but not
> wxPython-4.2.2-cp311-cp311-linux_x86_64.whl or
> wxPython-4.2.2-cp313-cp313-linux_x86_64.whl

Thanks for coming back on this. That is to be expected since the first wheel there is the only one matching your Python version.

> Is it sufficient to remove the venv folder?

Yes. The beauty of using venv is that you can start from scratch without messing with your computer installation. Removing that folder and following [the instructions][1] again should work. 

> In the initial installation i ran "python -m pip install ." before " python -m pip install https://extras.wxpython.org/wxPython4/extras/linux/*" , and in the working PC i did backwards. Can this cause this problem?

Also yes. Maybe is not emphasized enough on the note. First install the wxPython wheel, then run `python -m pip install .`. The reverse order defeats the whole purpose of installing that wheel separately.

[1]: https://github.com/kliment/Printrun?tab=readme-ov-file#running-from-source

lucianoloder, 2025-05-20:
Hi to all.

Much thanks for all the help, and sorry for the delay. I was chasing the wrong rabbit hole.

First i was thinking that the problem was that in the file "wx/core.py" it was importing a file called "._core", which was not present in the "wx" folder inside "site-paclages", and so i taught it was looking into my installation, causing the bug.

But now i understood that the "ImportError" is a linker problem, not python problem.
So, by executing "ldd" command i was able to see that "venv/lib/python3.12/site-packages/wx/_core.cpython-312-x86_64-linux-gnu.so" shared library is searching my system, not the venv environment:

[image]

The correct WXU_3.2.6 is indeed inside site-packages, but the linker is not searching there.

I can correct it by setting the LD_LIBRARY_PATH shell variable to venv: 

export LD_LIBRARY_PATH="$(pwd)venv/lib/python3.12/site-packages/"; python -c "import wx; print(wx.__version__);"

I used a relative PATH to LD_LIBRARY_PATH, and was able to open pronterface. Did not test it further, my 3d printer is not with me in this moment.

This variable can be set in the venv/bin/activate , and it may be the source of other bugs. Dont know how it would behave in a windows environment.

Thanks all for the help, and hope I could help with this amazing opensource project. Any further help i can give will be my pleasure.

DivingDuck, 2025-05-20:
Hi,
thanks for testing and reporting your results. I guess this will help other Linux users. It is maybe a good idea reporting your findings to the wxpython team as well.

> This variable can be set in the venv/bin/activate , and it may be the source of other bugs. Dont know how it would behave in a windows environment.

wxpython behaves in a windows environment fine. I run Pronterface on Windows10 with Python 3.13.3 and wxpython 4.2.3 from source in venv or compiled as executable w/o problems