kliment/Printrun #1485: Logging to stdout create a logging error and infinite loop

DivingDuck, 2025-02-25:
That one is working fine now, but found another one:

I got an logging error when Log to console is activated. I did a connect and disconnect. When I then try to close the Pronterface Window I'm not able to do so and got the second traceback w/o been able to close Pronterface. I need to kill the process manually.
Maybe also a path problem with dirs as I can't see any update in the log file.

**Edit**: I forgot to mention that the errors happen only if I run from the binary file - running from source is fine.

Terminal output:
```
Verbinde...
--- Logging error ---
Traceback (most recent call last):
Traceback (most recent call last):
Keine Verbindung zum Drucker.
--- Logging error ---
Traceback (most recent call last):
  File "logging\__init__.py", line 1153, in emit
  File "printrun\pronterface.py", line 113, in write
AttributeError: 'NoneType' object has no attribute 'write'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "printrun\printcore.py", line 171, in logError
  File "printrun\pronsole.py", line 284, in logError
  File "logging\__init__.py", line 2162, in error
  File "logging\__init__.py", line 1548, in error
  File "logging\__init__.py", line 1664, in _log
  File "logging\__init__.py", line 1680, in handle
  File "logging\__init__.py", line 1736, in callHandlers
  File "logging\__init__.py", line 1026, in handle
  File "logging\__init__.py", line 1158, in emit
  File "logging\__init__.py", line 1074, in handleError
  File "printrun\pronterface.py", line 113, in write
AttributeError: 'NoneType' object has no attribute 'write'

--- Logging error ---
Traceback (most recent call last):
Traceback (most recent call last):
```

_Originally posted by @DivingDuck in https://github.com/kliment/Printrun/issues/1484#issuecomment-2676930848_

DivingDuck, 2025-02-25:
@neofelis2X, 

Can you please check this as well for macOS binaries? I want to know if this is only related to the Windows binaries. Running from source do not create this issue 

**<ins>Test procedure:</ins>**

1. Start the actual development binary.
2. Activate via -->Settings -->Options -->User Interface `Log to console` and close the Window with ok.
3. Close and then restart Pronterface again
4. Connect to a printer and then disconnect.
5. Try to close Pronterface

Thanks in advanced.

neofelis2X, 2025-02-26:
Okay, i followed your instructions, made a binary from master, set `log to console` etc, also startet it `--verbose` to see if there are any extra messages. But no errors. Connects, printer moves, disconnects, pronterface closes without any complaints. Everything very normal.

DivingDuck, 2025-02-27:
Thanks for testing. So it seems like this is a Windows only issue.