kliment/Printrun #1488: Malware found in executable

TanktheShredder, 2025-03-18:
Downloaded latest windows release https://github.com/kliment/Printrun/releases/download/printrun-2.2.0/printrun-2.2.0_windows_x86_py3.10.zip

1. Malwarebytes found "Malware.Heuristic.2127" in PRONTERFACE.exe
2. Scanned the pronterface executable in VirusTotal, 22/72 security vendors flagged this file as malicious

[image]

[image]

kliment, 2025-03-18:
The executable does not contain malware. It contains a python interpreter, which is used in lots of known malware, and therefore the heuristic-based scanner misdetect it as such. We and other python-based projects frequently encounter this issue and there's not much we can do about it.

kliment, 2025-03-18:
See also: https://github.com/kliment/Printrun/issues/1159 https://github.com/kliment/Printrun/issues/1475 https://github.com/kliment/Printrun/issues/1446 and https://github.com/kliment/Printrun/wiki/FAQ

rockstorm101, 2025-03-21:
Duplicate of #1446

closed 2025-03-21 by rockstorm101