kliment/Printrun #1490: ImportError: cannot import name 'GL_LIGHTING' from 'pyglet.gl'

vido89, 2025-03-25:
Im on LM22.1, I 3d view does not work, I tried Printrun 2.2.0 na master

```
3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/slobodan/Printrun-master/printrun/gui/viz.py", line 69, in __init__
    import printrun.gcview
  File "/home/slobodan/Printrun-master/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/slobodan/Printrun-master/printrun/gl/panel.py", line 29, in <module>
    from pyglet.gl import glEnable, glDisable, GL_LIGHTING, glLightfv, \
ImportError: cannot import name 'GL_LIGHTING' from 'pyglet.gl' (/usr/lib/python3/dist-packages/pyglet/gl/__init__.py)
```

rockstorm101, 2025-03-25:
Hi @vido89, this is a known issue due to Printrun still relying on pyglet < 2. It should work normally if you just use the 2D view.

The request for migrating the code to pyglet > 2 is already tracked at #1468.

vido89, 2025-05-10:
Hi @rockstorm101 OK tnx for info, I switched to Debian 12 and it works, just needed 

`sudo apt-get install python-wxtools python3-appdirs python3-serial python3-pyglet python3-psutil`

rockstorm101, 2025-07-23:
Hi, I will close this issue since the root cause is tracked at #1468.