kliment/Printrun #1492: Locking Interface on DaVinci miniMaker

PlasmaKirill, 2025-04-10:
When I connect to printer - all is ok. I can move all axis, can extrude. But If I load a file gcode from program to print - my printer say "Locking Interface". And i can't do anything!

[image]

rockstorm101, 2025-04-12:
Did you try unchecking the option Settings > Options > User Interface > Lock interface upon print start ?

rockstorm101, 2025-04-19:
> > Did you try unchecking the option Settings > Options > User Interface > Lock interface upon print start ?
> 
> yes, I found it, but printer still not print. :(

Once the "Lock interface upon print start" and "Display interface lock checkbox" options are disabled. What's the error upon print? Maybe try enabling "debug communications" as well to get more info.

PlasmaKirill, 2025-04-19:
> Once the "Lock interface upon print start" and "Display interface lock checkbox" options are disabled. What's the error upon print? Maybe try enabling "debug communications" as well to get more info.

I havent a error! I made it - my printer was lock for printing spool with filament with NFC-TAG only!!!! I firmware him to a new S/N and now all work fine!
Thank you!

rockstorm101, 2025-04-19:
Great news! Thanks for reaching out and happy printing

closed 2025-04-19 by rockstorm101