kliment/Printrun #1493: setuptools required to build, but is not instralled by default on newer Debian

VanessaE, 2025-04-12:
I recently reinstalled my system in order to upgrade to Debian testing/"Trixie" (as "Bookworm" was gradually degrading from "stable" to "stale" and some stuff I rely on had quit working, as is so often the case with Debian), and for whatever reason, Printrun didn't want to build, until I found that `setuptools` doesn't get installed by default.

A simple `python3 -m pip install setuptools` after doing the `requirements.txt` step resolved it, after which `setup.py build_ext --inplace` was able to do its job normally.

This suggests to me that `setuptools` should either be added to `requirements.txt`, or `setup.py` should explicitly check for and install it when needed.  Or at the very least, the need for it should be mentioned in the README.

Since Debian testing doesn't give the OS itself a version number, I'll have to just say that at the time this happened (a bit less than two weeks ago), I had just finished the OS install maybe an hour or two prior.

This system uses Python 3.13.2.

Here's the full log of a successful build/install, with `setuptools` added in where I installed `Cython`, for reference:

<details><summary> Show the full log... </summary>

```
vanessa@rainbird:~$ # sudo apt -y install python3-wxgtk4.0 python3-pyglet cython3 python3-psutil python3-serial python3-platformdirs
vanessa@rainbird:~$ # already installed earlier :P
vanessa@rainbird:~$ cd /home/vanessa/RepRap/
vanessa@rainbird:~/RepRap$ rm -rf Printrun ~/.cache/pip/
vanessa@rainbird:~/RepRap$ git clone https://github.com/kliment/Printrun.git
Cloning into 'Printrun'...
remote: Enumerating objects: 13872, done.
remote: Counting objects: 100% (579/579), done.
remote: Compressing objects: 100% (160/160), done.
cd Printrunbjects:  26% (3687/13872), 28.18 MiB | 28.17 MiB/s
remote: Total 13872 (delta 512), reused 419 (delta 419), pack-reused 13293 (from 4)
Receiving objects: 100% (13872/13872), 81.81 MiB | 32.49 MiB/s, done.
Resolving deltas: 100% (8665/8665), done.
vanessa@rainbird:~/RepRap$ cd Printrun
vanessa@rainbird:~/RepRap/Printrun$ python3 -m venv venv
vanessa@rainbird:~/RepRap/Printrun$ source venv/bin/activate
(venv) vanessa@rainbird:~/RepRap/Printrun$ python3 -m pip install -r requirements.txt
Ignoring pillow: markers 'sys_platform == "win32"' don't match your environment
Ignoring pyobjc-framework-Cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyreadline3: markers 'sys_platform == "win32"' don't match your environment
Collecting pyserial>=3.0 (from -r requirements.txt (line 1))
  Downloading pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting wxPython>=4.2.0 (from -r requirements.txt (line 3))
  Downloading wxPython-4.2.2.tar.gz (57.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.4/57.4 MB 38.0 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy>=1.8.2 (from -r requirements.txt (line 4))
  Downloading numpy-2.2.4-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (62 kB)
Collecting pyglet<2.0,>=1.1 (from -r requirements.txt (line 5))
  Downloading pyglet-1.5.31-py3-none-any.whl.metadata (7.6 kB)
Collecting psutil>=2.1 (from -r requirements.txt (line 6))
  Downloading psutil-7.0.0-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (22 kB)
Collecting lxml>=2.9.1 (from -r requirements.txt (line 7))
  Downloading lxml-5.3.1-cp313-cp313-manylinux_2_28_x86_64.whl.metadata (3.7 kB)
Collecting platformdirs (from -r requirements.txt (line 8))
  Downloading platformdirs-4.3.7-py3-none-any.whl.metadata (11 kB)
Collecting puremagic (from -r requirements.txt (line 9))
  Downloading puremagic-1.28-py3-none-any.whl.metadata (5.8 kB)
Collecting dbus-python>=1.2.0 (from -r requirements.txt (line 10))
  Downloading dbus-python-1.4.0.tar.gz (232 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting six (from wxPython>=4.2.0->-r requirements.txt (line 3))
  Downloading six-1.17.0-py2.py3-none-any.whl.metadata (1.7 kB)
Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
Downloading numpy-2.2.4-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.1 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.1/16.1 MB 38.3 MB/s eta 0:00:00
Downloading pyglet-1.5.31-py3-none-any.whl (1.1 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 32.9 MB/s eta 0:00:00
Downloading psutil-7.0.0-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (277 kB)
Downloading lxml-5.3.1-cp313-cp313-manylinux_2_28_x86_64.whl (5.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.0/5.0 MB 36.4 MB/s eta 0:00:00
[... 37 lines cut ...]
/home/vanessa/RepRap/Printrun/venv/lib/python3.13/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: GNU General Public License v3 or later (GPLv3+)

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
running build_ext
Compiling printrun/gcoder_line.pyx because it changed.
[1/1] Cythonizing printrun/gcoder_line.pyx
building 'printrun.gcoder_line' extension
creating build/temp.linux-x86_64-cpython-313/printrun
x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/home/vanessa/RepRap/Printrun/venv/include -I/usr/include/python3.13 -c printrun/gcoder_line.c -o build/temp.linux-x86_64-cpython-313/printrun/gcoder_line.o
creating build/lib.linux-x86_64-cpython-313/printrun
x86_64-linux-gnu-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-x86_64-cpython-313/printrun/gcoder_line.o -L/usr/lib/x86_64-linux-gnu -o build/lib.linux-x86_64-cpython-313/printrun/gcoder_line.cpython-313-x86_64-linux-gnu.so
copying build/lib.linux-x86_64-cpython-313/printrun/gcoder_line.cpython-313-x86_64-linux-gnu.so -> printrun
(venv) vanessa@rainbird:~/RepRap/Printrun$ deactivate
cd ~
```

rockstorm101, 2025-04-13:
Hi @VanessaE, thanks for reporting this. Could try installing the "new way" and see if setuptools gets installed? I mean running `(venv) $ python -m pip install .` instead of the old `python3 -m pip install -r requirements.txt`. This "new way" should install both setuptools and Cython on its own because it uses `pyproject.toml` which lists both as build dependencies.

VanessaE, 2025-04-14:
Nope, that method is also not sufficient to bring in `setuptools` or `Cython`.

rockstorm101, 2025-04-19:
> Nope, that method is also not sufficient to bring in `setuptools` or `Cython`.

Thanks. That's weird. It was supposed to work like that. I'll try to reproduce this and come back to you.