kliment/Printrun #1494: Spanish Translation

ZombieEater, 2025-04-15:
Hi there! I wish to collaborate and make the spanish translation. Do the community think that whort it?
If yes, the I'll start translating.

Greetings!

#161

rockstorm101, 2025-04-19:
Hi @ZombieEater, thanks for contributing! Translations are more than welcome! Spanish is a widely spread language so it'll sure be beneficial to have that translation. Thanks again for volunteering.

ZombieEater, 2025-04-19:
I'll be doing more today, hope to finish in a pair of days :-)

ZombieEater, 2025-04-19:
:-) I wish to test my partial translation but I can't find the language selector in the program O_o  (locale folder is ok, but can't change language) Help!

ZombieEater, 2025-04-29:
@DivingDuck thanks.... I'll continue with the translation to the point to be uploaded and reviewed. I expetected to see a "Language" dropdown menu, because my computer is in english, and I use all in english too...bu I'll create another user in my computer to change the settings and test it.
Greetings!