kliment/Printrun #1495: Avast flagging Printrun as malware. Again.

rv7charlie, 2025-04-15:
I know this isn't anything new, but it would be nice if we didn't have to deal with it with every download. I have no idea what the reason is, but it would be great if it could be fixed, or at least explained. Apparently this isn't just Avast doing this, since others have reported the same problem with different antivirus software.

The current message says, "We prevented your connection to objects.githubusercontent.com because it is a dangerous webpage. Threat category: Win64:Malware-gen.

I wish I could help, but my level of expertise with software and virus management is reporting problems.

Thanks for any assistance in clearing this up.

ZombieEater, 2025-04-15:
I used to make an exception for the folder where I put the program, so Avast ignore that folder.

Here's how: https://support.avast.com/en-us/article/antivirus-scan-exclusions/

rv7charlie, 2025-04-16:
With Avast, it's not the destination; it's the website. Avast won't even allow the download unless the website is 'whitelisted' (for lack of a better term). Worse, I'd already put this site in the whitelist when I downloaded the previous version of Printrun.

rockstorm101, 2025-04-19:
The issue is known. Not strictly Printrun-related. If anybody reading this has any suggestions on how to deal with this please speak up :)

Duplicate of #1176

closed 2025-04-19 by rockstorm101