kliment/Printrun #1499: gtk_box_gadget_distribute error on Linux Mint

NoxAstrum1, 2025-06-01:
I apologize if this isn't the right place to post this.

I've been using pronterface successfully since I switched to Linux. I recently switched distros, and then switched back to Mint. I installed Printrun the same way I did last time, but I think I screwed something up because I now get an error:

(pronterface.py:324176): Gtk-CRITICAL **: 19:21:02.160: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

The application window appears, but I get these errors in the terminal. It also now fails to auto-detect the correct port, and so can't connect to the printer. It's using /dev/ttyACM0. The printer is connected to a USB port.

Could this be a result of having Python3 installed instead of Python, or could it genuinely be a bug?

DivingDuck, 2025-06-02:
> I installed Printrun the same way I did last time,

How did you install it? Python should be something between Py 3.8 and 3.13. Don't use Py 2.x as this isn't supported since a very long time starting with Pronterface 2.0. 

Usually this happen when some of the libraries are not in the version available they need to be. One is wxPython (>4.2) another is pyglet (<2,0) or any of the other needed libraries. 

Have you follow the guide on our start page?
https://github.com/kliment/Printrun?tab=readme-ov-file#linux-packages-from-official-repositories and 
https://github.com/kliment/Printrun?tab=readme-ov-file#running-from-source

Running from source you need to install some additional libraries you may not have installed by default in the virtual environment before installing the libraries from requirements.txt:
setuptools, cython, simplejson, pyinstaller, polygon3

@rockstorm101 as I know I'm not a Linux guy, can you maybe give him a helping hand?

rockstorm101, 2025-06-02:
> Could this be a result of having Python3 installed instead of Python, or could it genuinely be a bug?

Definitely not the first. Might be a bug, though I'm unable to reproduce it at the moment.

Please tell us how you installed Printrun and what versions of each dependency you run.

rockstorm101, 2025-06-07:
Hi @NoxAstrum1, thanks for coming back on this. Glad you fixed the port issue.

Regarding the "gtk_box_gadget_distribute" error. It feels like a wxPython error.
 1. Could you try installing the system package `python3-wxgtk4.0`?
 2. If that alone does not solve it. Please try using the "--system-site-packages" when creating you venv instead of installing the Ubuntu wheel from "https://extras.wxpython.org" to see if using the system package solves the gtk issues.

On option 2 above I mean the following process:
1. sudo apt install git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0-dev python3-wxgtk4.0
2. git clone https://github.com/kliment/Printrun.git
3. cd Printrun
5. python3 -m venv --system-site-packages venv
6. source venv/bin/activate
7. python -m pip install .
8. ./pronterface.py

NoxAstrum1, 2025-06-11:
> rockstorm101>  left a comment > (kliment/Printrun#1499) <https://github.com/kliment/Printrun/issues/1499#issuecomment-2952944084>

I tried step 1, but it didn't work. I then tried step 2, it didn't work either. If it's just something I did wrong during the installation (which I fully believe) and not a bug, I can just try a fresh install. It's not stopping me from printing, so I wouldn't waste your time on it.

VanessaE, 2025-06-22:
I have been pinged! 🙂

Python is such a mess anymore... if it were up to me, I'd see Printrun/Pronterface rewritten in C and merged into SuperSlicer to replace the old control panel that its predecessor Slic3r once had.  Slice and print from one program, without the hassle. 🙂